iceman1001/ChameleonMini-rebooted #79: NFC-list dose not show my tag

ictmehrdad, 2019-03-04:
Hi
About the reader attack with chameleon mini e rebooted, In MF detection and with mfkey32 I extract the key B and I download the DUMP and go to the Kali, when I enter the NFC-List it just show:
nfc-list uses libnfc 1.7.1
NFC device: SCM Micro /SCL3711-NFC&RW opened
But in a video of Lab-401 I saw that it shows something else 
When I use Mfoc command :
The custom key 0*xxxxxxx has been added to the default key
mfoc: ERROR: No tag found

I connect both the SCL3711 and chameleon to my kali os
What should I do!?
Thanks

ictmehrdad, 2019-03-04:
[image]
[image]
[image]

iceman1001, 2019-03-04:
I dont understand the issue for the chameleon firmware?  What did you do on the chameleon and what doesnt work with the cameleon firmware?

ictmehrdad, 2019-03-05:
Thanks for your attention 
I mean I put my chameleon in MF detection and set a valid UID on the slot one then I go to the Reader then put it on the reader for 5 sec then I use the mfkey32 to get the key and I go the key B
Then I download the DUMP file and go to the kali Linux and connect both chameleon and SCL3711 to the kali os and use the nfs-list and mfoc commands that you saw in the last post! But Nothing happened!
What should I do!?

ictmehrdad, 2019-03-09:
Dear iceman
Could you please help me!?
Thank you

iceman1001, 2019-03-09:
so you used chameleon mini with **MF_DETECTION** and it work since you got a key.
..then you downloaded a dump?   From where?  Your chameleon doesn't have a valid dump on it if you are in MF_DETECTION...    You will now need to use the key you found against the **card** to get all other keys and be able to make a proper dump from the card.    So this is no issues here.  Please ask your question about libnfc in their forum instead.    If you need to understand what to do with a key, you should read up on Mifare Classic datasheets and look at some videos about mifare hacking and/or read some blog posts about it.  

Good luck!

closed 2019-03-09 by iceman1001