iceman1001/ChameleonMini-rebooted #84: Samsung door lock has no response to ChameleonMini

soffchen, 2019-03-12:
I copied my samsung key to ChameleonMini slot 1, flashed with the latest firmware under iceman fork. Everything read by PM3 is the same at all. But when I place ChameleonMini on the card reader of the lock, the led flash once, and the lock has no response. I also write the dump to an empty id card, it can open the door successfully.

Any idea? ps, the original firmware has the same issue.

soffchen, 2019-03-12:
card type matches

```
--> ATQA?
101:OK WITH TEXT
0004

--> SAK?
101:OK WITH TEXT
08
```

sak/atqa set correctly?

```
--> LOGMODE?
200:UNKNOWN COMMAND

--> LOGMODE?
200:UNKNOWN COMMAND

--> LOGMODE
200:UNKNOWN COMMAND

--> LOGDOWNLOAD
200:UNKNOWN COMMAND
```
 log functions not implemented yet?

iceman1001, 2019-03-12:
the sak/atqa depends on what your dump files say and also what your lock wants.   hence the logtrace..

Ollibolli85, 2019-05-07:
I have the exact same problem!

shinhub, 2019-08-14:
@Ollibolli85 , @soffchen , please try with new compiled firmware as of today, a lot have been reworked on Mifare (I guess you talk about Mifare cards). As a reminder, this repo and firmware is RevE rebooted only. If you still have an issue, please do describe the testing scenario.

iceman1001, 2019-08-18:
@soffchen   @Ollibolli85    any news?  or is it time to close`?

Ollibolli85, 2019-08-19:
Hello iceman1001,

sorry for the late answer!

I will test the new compiled firmware the next days!

soffchen, 2019-08-19:
> @Ollibolli85 , @soffchen , please try with new compiled firmware as of today, a lot have been reworked on Mifare (I guess you talk about Mifare cards). As a reminder, this repo and firmware is RevE rebooted only. If you still have an issue, please do describe the testing scenario.

Still no response:

```
I copied my samsung key to ChameleonMini slot 1, flashed with the latest firmware `7420671`. Everything read by PM3 is the same at all. But when I place ChameleonMini on the card reader of the lock, the led flash once, and the lock has no response. I also write the dump to an empty id card, it can open the door successfully.
```

Wait @Ollibolli85 's result.

shinhub, 2019-08-19:
@soffchen it strange cause I still have such issues as well with a doorlock. This will require some sniffing of the communications, with a working card and with the Chameleon, if anyone can.

Ollibolli85, 2019-08-19:
Also not working here!

shinhub, 2019-08-19:
@Ollibolli85, @soffchen could you please try a thing for to narrow things down. Getting sure you have last firmware and that you program with rebootedGUI, what happens with your door lock if you put a real other card (a blank one would be ideal, but really any card, except of course the one that is actually supposed to open the door...) in front of your Chameleon and try again?

jimbauwens, 2019-09-07:
Hi, I have a similar issue ( with the latest firmware, have not tested other versions ). 

The device properly opens one lock, but not another one (different type). When placing another card (blank/not the actual card) behind the chameleon, it sometimes does work.

With not working, I mean that the lock / reader does not even detect that anything is placed before it (if it doesn't like the card, it blinks in error).

shinhub, 2019-09-11:
@jimbauwens thanks confirming the blank-card override trick, that also worked for me most of times : )
This however did not work at all with older firmwares.

I wonder if this may be a power/timing issue with some readers. Could not identify the root cause, despite some sniffing efforts on those case, so I guess the root cause is not at Application side, but more at hardware or codec side.

iceman1001, 2019-09-14:
Hard to debug based on readers / phones,   we can't replicate the problem easily and we can't exclude that its bad reader / phone either.   I am closing this one.

Final advice,  try latest code,  @shinhub  did some impressive modifications.

closed 2019-09-14 by iceman1001

shinhub, 2019-09-14:
@iceman1001 thanks again, but for this one I really think there is something we can change. This happens to me in multiple case, plus MFC quite "difficult" reading sometimes, and I think it might be due to RSSI/Antenna levels settings. As I have access to such a reader where ChameleonMini is not read at all, I think I may attempt some blind macro constants and settings change to AntennaLevel, and see what happens. Up to you though.

iceman1001, 2019-09-14:
Its up to @soffchen to try latest source code and start a new issue if problem still persists.  Can't have open stale issues with no feedback from OP.

szjiajin, 2022-12-30:
> @iceman1001 thanks again, but for this one I really think there is something we can change. This happens to me in multiple case, plus MFC quite "difficult" reading sometimes, and I think it might be [...]

Hello, I have had the similar problem. The reader does not correctly identify the my chameleon card while pm3 can communicate without any problem. I tried to sniff the protocol between card reader and chameleon using pm3. It seems that the card reader kept sending the Wakeup command and the card response with 04 00 sometimes. I guess maybe it is because the card reader did not receive the response from chameleon or the latency is too high. 
I also tried to log the communication by using log support in chameleon but I didn't know how to properly use it.
As I am not familiar with chameleon firmware, if you still want to dig deeper into this issue I can provide my work with more details. Thank you!

szjiajin, 2023-01-03:
> Hello, I have had the similar problem. The reader does not correctly identify the my chameleon card while pm3 can communicate without any problem. I tried to sniff the protocol between card reader [...]

I have solved my problem. Just to give some update on my problem if anyone cares: I disassembled the card reader and used a wire that is too long to connect it to the antenna. This affects the communication in some way. However, this modification only affect chameleon and pm3, for an actual Mifare card communication can still be carried out normally. I am no electrical engineer so I have no idea what was the actually going on, but I hope my description helps.