iceman1001/ChameleonMini-rebooted #86: MF Classic Emulation BROKEN

eXspir3, 2019-03-12:
Hi :)

So i was trying to emulate 2 different kinds of MFC 1k today and sadly had to see that it is **not working realiably AT ALL**
I did the following:

1. Upgrade Firmware to Iceman 1.3
2. Upload .mfd dump file to slot, choose MFC 1K & hit apply

--> When presenting the Reader with the Card it does not react
turns out when trying to read the chameleon with my smartphone it showed correct UID SAK and ATQA but sectors where randomly seen as unreadable - sometimes reading sectors succeeded but never all of them
--> For my usecase the reader must be able to read the chameleon correctly in order to work as parts for the key generation of the reader are in some sectors (not only UID)

Might this be a timing issue or has anyone resolved it?
**_Sectors seem to be randomly readable or unreadable_**

Hope someone can help me 😁

Best regards

iceman1001, 2019-03-12:
Did you use the GUI for uploading?

eXspir3, 2019-03-12:
Yes i used your newest Version of Chameleon GUI :/

iceman1001, 2019-03-12:
ok,  verified.    The gui isn't the problem. 
using mini and pm3. 
On latest firmware the 14a is slow and acting strange.  can do **hf mf fchk** to get all keys, but the dumping doesn't work.

revert to stock firmware and **hf mf fchk** / ** hf mf dump**  works good.

eXspir3, 2019-03-12:
Ok thanks i will try - where do i get the stock Firmware? / Would you be able to provide me with the .bin or .eep .hex files?

eXspir3, 2019-03-12:
Sadly using Stock Firmware (provided in the GUI) did not resolve the problem

Best Regards

shinhub, 2019-08-14:
@eXspir3 you should try with firmware compiled from today. A lot have been fixed. Note that some readers still do not react, but if this is still the case for you, you should be able to narrow it to a test case you could describe.

iceman1001, 2019-08-17:
time to close?

eXspir3, 2019-08-17:
unfortunately i cant test right now - pls close

closed 2019-08-17 by iceman1001