iceman1001/ChameleonMini-rebooted #87: MIFARE 1K : reader attack fails authentication

BackScatt, 2019-03-13:
I have a very strange issue with the Chameleon Rebooted : the emulation of a MIFARE 1K works great but I always have authentication problem with the reader attack.

First of all, the config: 
* the reader is a PN532 module connected via I2C to a raspi
* to read a card I use a tiny python script with a python lib for the PN532
* I use the latest GUI available (by the way resizing is very helpful)
* I compiled the latest firware (without xxxMY for the commands) on linux with the makefile and then upgraded the Chameleon from windows with the .bat file.

What I've done : 
* I dumped an empty card with the script and the PN532 : everything worked great, authentication for the 16 sectors, I can read all the blocks.
* I configured the slot1 for 1K emulation and loaded the dump. I ran the same script, worked great as well, exact same dump.
* I configured the slot2 for reader-attack and loaded the dump. I ran the first script but it raise authentication error (ie: PN532 in able to authenticate with 1K emulation but not reader-attack) AND when I click the 'mfkeys' GUI button I've got the message saying that there is not key (I tried several things like with/without clicking 'refresh' before 'mfkeys').

Does anyone have a clue for this issue ?

PS : I've also seen that there was a problem using dfu-programmer under linux to flash the uc, is that issue fixed or still under investigations ?

bogiton, 2019-03-15:
Hey @BackScatt 
The MF_DETECT configuration isn't supposed to do a proper authentication sequence. It will just store the nonces when the reader tries to authenticate with it. You only need those in order to run an offline key recovery. After acquiring the proper key(s), you can configure the slot to MF_CLASSIC_1K, upload a full dump and then emulate properly against the reader.
As for the mfkey32 button that it's not working, have you got the latest versions of both the firmware and the GUI?
P.S. Flashing with just the dfu-programmer is getting closer to be accomplished, as you can see in issue #11, thanks to @slurdge

BackScatt, 2019-03-16:
Oh I thought that it will properly respond to each authentication tried by the reader and log the key for each attempt. That means that this attack can only recover 1 key ?
And yes I double checked, the firmware is the one without the 'my' for the serial commands and same thing for the GUI which is resizable so both ones are up to date.

iceman1001, 2019-03-17:
I think there is space for 6 possible keys in the MF_DETECTION.

eXspir3, 2019-03-18:
@BackScatt @iceman1001 

Also thought i would be able to log the whole traffic - do you think it is possible with this hardware?

yaxax, 2019-04-07:
I have the chameleon mini reve when i do mf detect and then in the uid section i put my uid and then i do switchcard  then apply but then the uid section goes back to ffffffff why that happens?But when I do mf classic 1k and put my uid in the uid section off the gui then the uid goes good and don't goes back to ffffffff when I press apply what is that?

yaxax, 2019-04-25:
Can someone explain to me why the mf_detection always goes back to ffffffff. Why I ask this question because I dont get support from lab 401, they never answer your questions about the products you buy from them. I bought a second one  again at Lab401 and also had the same problem.The mf_detection again goes back to ffffffff Personally I never gonna buy anything from Lab401 again. But rysc corp you always get support from Nick.And is because Rysc Corp don't sell the chameleon mini reve rebooted  because if they did i bought it their just because off the support. So thats why i ask this question about mf_detection why the uid always goes back to ffffffff thanks.

yaxax, 2019-05-11:
When i do the reader attack whit the chameleon mini reve rebooted do I need the card to go further are can i go further whitout having the card to open the reader lock

bogiton, 2019-05-17:
@yaxax It depends on the reader implementation. It may require more data (from the card) and not just the UID.

closed 2019-05-18 by iceman1001