iceman1001/ChameleonMini-rebooted #88: Getting started on Windows 10 : beginners walkthrough

shinhub, 2019-03-23:
Quick insights for RevE beginners as I am really new to all this, and it took me some hours figuring it all. Some directions here may be added to wiki (I don't know how to git, sorry).

First of all, do Windows before anything else if your goal is to get RevE working fast. It is the easiest way. On Linux and MacOS, the Qt based GUI did not work each time from me, flashing has been a pain, and temped to build firmware, I have been burning time on some traditional compilation and environment issues. Working on a Windows VM is possible, it worked just as well for me.

Just **check your device can be talked with** on Windows 10. This should be plug & play. Plug your device in USB port, and use a GUI tool (had the better results with [this one from Lab401](https://lab401.com/products/chameleon-mini/software.zip), based on this repo). The output console of GUI Tool should say `Success, found Chameleon Mini device on 'COM3' with Firmware RevE rebooted installed`

**Put your RevE in boot loader mode**, directly from the GUI (Settings tab, then "Upgrade"). Your RevE should be setup as a new unknown device.

**Setup the DFU drivers** so that device is correctly recognized. To do so in Windows 10, get the [Drivers files from this repo](https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Drivers) (you can get all files from this repo by choosing "Clone or Download" then "Download ZIP" on the [main repo page](https://github.com/iceman1001/ChameleonMini-rebooted)), then in Windows type "Device manager" in start menu, open it, right click on unknown USB device, and choose "Update driver". Then points the assistant to the folder with "Drivers" from this repo, and it will be installed. You should then see your USB device as "Atmel USB Devices" ATxmega32A4U.

**Get existing firmwares**, or build a new one from source : 
- Get existing one : [original manufacturer one](https://github.com/iceman1001/ChameleonMini-rebooted/files/1573135/bootloader-files.zip) (the one built-in when you purchase), or [latest manufacturer compiled one](https://lab401.com/products/chameleon-mini/firmware/latest.zip) (could be newer)
- Build your one from this repo : managed to to it on Linux (only), [as described on this repo](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing-Linux-(Unix)) Wiki (except I had to tune idProduct in udev rules, according to lsusb command result, as mine was not the same than in shipped file)

**Setup Windows C++ Redistributable for Visual Studio 2013** (VC 12.0). If you use a 64bits Windows, you ALSO have to setup the x86 ones (I had a hard time on that, the flash utility does not start if you do not have the x86 VC 12.0 redistributable on a 64bits system). Both (x64 and x86) can be downloaded [from Microsoft here.](https://support.microsoft.com/en-us/help/3179560).

**Flash your firmware** with "BOOT_LOADER_EXE.exe", that you will already have in the "Extras" folder of the GUI tool you downloaded earlier. You just have to put the "myfile.bin" and "myfilee.bin" files you downloaded as ZIP archive from links in "Get existing firmwares" in the same folder, and double click the "BOOT_LOADER_EXE.exe". Detailed instructions, including case where you compiled your firmware (an extra step and more tools from the "Software" dir in this repo is necessary, to pack .hex and .eep files as .bin. You will use "flash.bat" directly and only - it will fire up BOOT_LOADER_EXE by itself at the end), can be found [on this repo wiki](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing---windows).

Close GUI, disconnect USB RevE, connect it back and fire the GUI up once again. Now **you should be all set**. 
- From now (?) RevE **can only get a first key from a dialogue with a reader, randomize/set UIDs, and emulate various MIFARE cards** from dumps you made with another reader device (proxmark, or cheap SLC3711 / ACR122u for instance). 
- Each slot (8 slots total) can be setup with some functionality (reader sniff, or card copy emulation, or UID player). You have to select the slots you want to set in the GUI, tune settings, then click "Apply". "Upload Dump" and "Download Dump" act on current "active" slot, that you can choose with "Set Active".
- Slots are numbered from 1 to 8 in GUI, but from 0 to 7 in command line.
- You can only do MIFARE 4k on 1st slot.
- The "Dump Management" tab is useful to open cards dumps and change them if needed. Use a "Template" to highlight cards keys and ACL. If you open a dump here, then you won't be able to upload the same file in a slot while loaded. Save your opened dump in another file, or close and reopen GUI.

Commands reference will be useful if you want to check or get ready to set your RevE from a command line terminal. [It is all here on this repo wiki](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Terminal-Commands). Just note that for any linked firmware in this issue, you must add "MY" at the end of each command. (i.e. commands names all end with "my"). If you compiled the firmware by yourself from this repo recently, you should not have the "my" suffix.

You will get some more beginners info from #36.

shinhub, 2019-03-23:
If you happen to completely "brick" the device (it does not light at all anymore, but is still can be plugged and recognized as a ATxmega32A4u USB device), playing with compiled firmwares, dfu-programmer, and/or avrdude on all platforms, like I did and like you probably will:
- go back to Windows, it all started here...
- in the GUI set from the 1st link at Lab401, get the sane "Extras" folder
- add the tools from this repo at directory Software/FlashTools in this "Extras" folder
- add the .hex files from this repo at directory Firmware/compiled in this "Extras" folder
- rename "ChameleonMiniRDV2.0_ATxmega32A4U.hex" as "Chameleon-Mini.eep"
- rename "ITS_A_CARD.hex" as "Chameleon-Mini.hex"
- if you have any "libusb0.dll" in the "Extras" folder, move or rename it so that it is not loaded
- go with "flash.bat", and you should get a bootloader mode ready blank Chameleon
- now go with a proper firmware flashing, and it should be allright again.

iceman1001, 2019-03-24:
yup,  the how-to guides that doesn't exist should have included this :)
There is a german pdf version,   but if you find time to make a Wiki entry that would be great.

If you want the latest version of the GUI,  https://github.com/iceman1001/ChameleonMini-rebootedGUI

slurdge, 2019-03-24:
@shinhub Thank you for the how-to guide. I have one question, did you buy the hardware on Lab401 or on other website ?
Also, if you need help to put it in the wiki, I can help you (or if you're fine with it, edit it and put it for you).

shinhub, 2019-03-24:
Would definitely fit in the Wiki, but was not quite sure it would fit any "editorial" standards first, and I do not know how to git (at all: commit, requests and so) + new to GitHub then. If it seems OK for you to start with that, and if I can edit Wiki without messing with the repo sources (it seems I just have to "edit" on the web page), then I will go with that! Do you want me to put it somewhere in particular, or do I go with like a "Getting started" page?

@slurdge yes bought on Lab401, but as far as I understand, there is a whole network of providers selling the same one ("ProxGrind" labelled).

iceman1001, 2019-03-24:
Here you go,  just edit this page  https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Getting-started

shinhub, 2019-03-24:
Great, I also messed with a Sidebar in an attempt to organize content. I think it helps with readability. If you don't like it you can simply remove it (flat pages list is still available). I will integrate this content.

shinhub, 2019-03-24:
Done!

closed 2019-03-24 by shinhub