iceman1001/ChameleonMini-rebooted #89: Compilation fails on Mac OS X Mojave (10.14.3)

shinhub, 2019-03-23:
I have been trying to **compile firmware from this repo on OS X Mojave with Homebrew** setup but it failed.

I have been following instructions from this [repo Wiki home page on Mac OS X](https://github.com/iceman1001/ChameleonMini-rebooted/wiki#mac-osx), and [checking with this ebfe.pw guide on Mac OS X](https://ebfe.pw/chameleon-mini-with-macosx-101/) to setup my environment.

Here is my environment : 
- OSX 10.14.3, `Darwin Kernel Version 18.2.0: Thu Dec 20 20:46:53 PST 2018; root:xnu-4903.241.1~1/RELEASE_X86_64 x86_64`
- Homebrew 2.0.5, `Homebrew/homebrew-core (git revision 05c9; last commit 2019-03-23)`, `Homebrew/homebrew-cask (git revision e2d7f; last commit 2019-03-23)`
- avr-gcc (GCC) 8.3.0 from osx-cross/avr tap
- git show : `commit 1fbd3ac99ae0f09abd7e39ff33b9a86bcfb007f6 (HEAD -> master, tag: untagged-f2bfc8d48745456bde59, origin/master, origin/HEAD)`

First try with `make `on ChameleonMini-rebooted/Firmware/Chameleon-Mini ends with this failed result : 
```
avr-gcc -c -pipe -gdwarf-2 -g2 -mmcu=atxmega32a4u -fshort-enums -fno-inline-small-functions -fpack-struct -Wall -fno-strict-aliasing -funsigned-char -funsigned-bitfields -ffunction-sections -I. -DARCH=ARCH_XMEGA -DF_CPU=32000000UL -mrelax -fno-jump-tables -x c -Os -std=gnu99 -Wstrict-prototypes -Werror -DUSE_LUFA_CONFIG_HEADER -DFLASH_DATA_ADDR=0x10000  -DFLASH_DATA_SIZE=0x10000  -DSPM_HELPER_ADDR=0x21FE0  -DBUILD_DATE=\"190323\" -DCOMMIT_ID=\"1fbd3ac\" -DCONFIG_MF_CLASSIC_1K_SUPPORT -DCONFIG_MF_CLASSIC_4K_SUPPORT -DCONFIG_MF_ULTRALIGHT_SUPPORT -DCONFIG_MF_DETECTION_SUPPORT -DSUPPORT_FIRMWARE_UPGRADE -DDEFAULT_CONFIGURATION=CONFIG_MF_CLASSIC_1K -DDEFAULT_BUTTON_ACTION=BUTTON_ACTION_CYCLE_SETTINGS -DDEFAULT_BUTTON_LONG_ACTION=BUTTON_ACTION_CYCLE_SETTINGS -DDEFAULT_SETTING=0 -DDEFAULT_PENDING_TASK_TIMEOUT=50  -DENABLE_EEPROM_SETTINGS  -I. -Isrc/LUFA/LUFA/.. -DARCH=ARCH_XMEGA -DBOARD=BOARD_NONE -DF_USB=48000000UL  -MMD -MP -MF Bin/MifareClassic.d Application/MifareClassic.c -o Bin/MifareClassic.o
Application/MifareClassic.c:191:22: error: 'abTrailorAccessConditions' defined but not used [-Werror=unused-const-variable=]
 static const uint8_t abTrailorAccessConditions[8][2] =
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
Application/MifareClassic.c:129:22: error: 'abBlockAccessConditions' defined but not used [-Werror=unused-const-variable=]
 static const uint8_t abBlockAccessConditions[8][2] =
                      ^~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Bin/MifareClassic.o] Error 1
```

This can easily be bypassed, removing the `-Werror` option on Makefile : 
- `cp Makefile Makefile.old`
- `sed -i -e 's/-Werror//g' Makefile`
- `make clean`
- `make`

But then I am stuck here at linking, and do not dare to mess more with env or files : 

```
[LNK]     : Linking object files into "Chameleon-Mini.elf"
avr-gcc Bin/Chameleon-Mini.o Bin/LUFADescriptors.o Bin/System.o Bin/Configuration.o Bin/Random.o Bin/Common.o Bin/Memory.o Bin/Button.o Bin/Settings.o Bin/LED.o Bin/Map.o Bin/AntennaLevel.o Bin/Uart.o Bin/Terminal.o Bin/Commands.o Bin/XModem.o Bin/CommandLine.o Bin/Codec.o Bin/ISO14443-2A.o Bin/MifareUltralight.o Bin/MifareClassic.o Bin/ISO14443-3A.o Bin/Crypto1.o Bin/Crypto1_asm.o Bin/Detection.o Bin/HIDParser.o Bin/ConfigDescriptors.o Bin/DeviceStandardReq.o Bin/Events.o Bin/HostStandardReq.o Bin/USBTask.o Bin/Device_XMEGA.o Bin/EndpointStream_XMEGA.o Bin/Endpoint_XMEGA.o Bin/Host_XMEGA.o Bin/PipeStream_XMEGA.o Bin/Pipe_XMEGA.o Bin/USBController_XMEGA.o Bin/USBInterrupt_XMEGA.o Bin/AudioClassDevice.o Bin/CDCClassDevice.o Bin/HIDClassDevice.o Bin/MassStorageClassDevice.o Bin/MIDIClassDevice.o Bin/PrinterClassDevice.o Bin/RNDISClassDevice.o Bin/AndroidAccessoryClassHost.o Bin/AudioClassHost.o Bin/CDCClassHost.o Bin/HIDClassHost.o Bin/MassStorageClassHost.o Bin/MIDIClassHost.o Bin/PrinterClassHost.o Bin/RNDISClassHost.o Bin/StillImageClassHost.o -o Chameleon-Mini.elf -lm -Wl,-Map=Chameleon-Mini.map,--cref -Wl,--gc-sections -Wl,--relax -mmcu=atxmega32a4u -Wl,--section-start=.flashdata=0x10000  -Wl,--section-start=.spmhelper=0x21FE0 
/usr/local/opt/avr-binutils/bin/avr-ld: Bin/LED.o: in function `LEDTick':
/ChameleonMini-rebooted/Firmware/Chameleon-Mini/LED.c:40: undefined reference to `LEDMode'
collect2: error: ld returned 1 exit status
make: *** [Chameleon-Mini.elf] Error 1
```

Could you please help me go further?

maxieds, 2019-03-27:
@shinhub 
I don't see a nice way around this, no. Have you considered filing a pull request into the repo?

shinhub, 2019-03-27:
@maxieds yeah, if only I knew how to git, and fill a pull request :)

maxieds, 2019-03-27:
It's really not that bad... You fork the repo, recommit the changes, and
then press a button to submit the pull request. Do you want me to do this,
or can you handle it from here?

shinhub, 2019-03-27:
@maxieds Ah ah, I secretly planned to try it anyway, but now I will be able to blame it on you if I break something :) Seriously. I will try, thanks!

iceman1001, 2019-03-27:
@maxieds   how about you do it ...  
However the suggested solutions isn't very good.   The -werror should still be there.   Correct code by comment out the arrays.

maxieds, 2019-03-27:
@Iceman1001
Sure. Let me get the avr gcc tools reinstalled on my remote Mac box and
then I will test and submit the PR.

shinhub, 2019-03-27:
@iceman1001 @maxieds yes did not want to remove -Werror either. Managing it by commenting out unused structs, and testing that it did not break anything on Linux also. But then there are more errors. Seems like default AVR GCC on OSX Homebrew "osx-cross" tap (8 and so) is not so permissive as the one I have on Linux (5 and so).
Next one is this one : 
```
src/LUFA/LUFA/Drivers/USB/Class/Common/HIDParser.c:93:5: error: 'memcpy' accessing 42 bytes at offsets 28 and 0 overlaps 14 bytes at offset 28 [-Werror=restrict]
     memcpy((CurrStateTable + 1),
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
            CurrStateTable,
            ~~~~~~~~~~~~~~~
            sizeof(HID_ReportItem_t));
            ~~~~~~~~~~~~~~~~~~~~~~~~~
```
Tried to update LUFA sources to latest ones, but still the same.
Maybe the good solution is to add "version 5" as a requirement for avr-gcc to compile :)

shinhub, 2019-03-27:
Patched LED.h with `void LEDMode(void);`
Then it's all because of LUFA...
Updated LUFA sources to 170418 from http://www.fourwalledcubicle.com/LUFA.php
Replaced the memcpy by memmove ("been there done that", [here](https://github.com/qmk/qmk_firmware/pull/2914/commits/f06765446eee680ebaeeb8feb13acb3e4a920f9b)).
Removed a bunch of "ATTR_CONST" each time compiler `said error: 'const' attribute on function returning 'void'`.
Then had to change some voids for `USB_ClassInfo_XXX_Device_t* const YYY`, add some stubs and fix some calls, because of "alias" related errors like: `'EVENT_Audio_Device_StreamStartStop' alias between functions of incompatible types 'void(USB_ClassInfo_Audio_Device_t * const)' ` ("been there done that", [here](https://github.com/qmk/qmk_firmware/pull/3510/commits/689e4ca0a949ec0a8c1a4d882930092536740e1d) - yeah, yeah, totally got that....).

Annnd it compiles like a charm, with `avr-gcc (GCC) 8.3.0` on OSX as well as `avr-gcc (GCC) 5.4.0` on GNU/Linux.
Was definitely a GCC 8 issue more than an OSX specific case.

maxieds, 2019-03-27:
Please go ahead and submit the pull request. I'm not going to have time to
get to it before this weekend.

iceman1001, 2019-03-28:
Great effort!  Everytime gcc gets an update it breaks and/or warnings about something.  Keeping up with having a correct build on different os/gcc is a pain in the butt.  Seen in pm3 repo. 

The PR is huge,  and feels like it could have been divided into several pr's.   As it is now it messes up with current usb enumeration,  changes code styles etc.   I prefer PR's that are smaller and does specific changes.   Making sure that this PR doesn't break anything elsewere ....

closed 2019-04-02 by shinhub