iceman1001/ChameleonMini-rebooted #90: Anyone in US/Canada with AVRisp programmer?

dealaddict, 2019-03-27:
My unit is bricked and I am not too confident about how to use the AVRisp programmer. The more I read the more I am getting confused about it. Is there anyone in us or canada who has it and willing to unbrick my device? I can pay for both ways shipping.

Please delete if its not appropriate to post here, its a long shot but I thought to ask away.

maxieds, 2019-03-27:
@dealaddict 
What exactly do you mean "*bricked*"? Do you mean that you cannot re-flash the RevE board with firmware? Have you tried [these tools](https://lab401.com/products/chameleon-mini/software.zip) yet from Windows 10? It doesn't seem like you should need to have someone else re-flash the device for you.

maxieds, 2019-03-27:
@dealaddict 
I have found that the *Lab 401* tools are sometimes better than the ones provided on this site even ... Can you try to recover the bricked device using that download source?

dealaddict, 2019-03-28:
@maxieds  yes its completely bricked this time. The last time I was able to use the software and reflash but now its completely dead, just a quick dim light flash upon connection before everything else goes dead.

slurdge, 2019-03-28:
@dealaddict is it a Lab401 device or Proxgrind? Did you try the button + plug in approach ?

dealaddict, 2019-04-10:
@slurdge  Its Proxgrind written on the back. I tried the button and plugin but it didnt work. I have the avr isp but it seems complicated

slurdge, 2019-04-11:
It's not that complicated. Pinout is in the wiki. Once connected, just fire AtMel Studio 7.0 and you can erase/flash the chip (you can even remove the bootloader if you fancy it).

dealaddict, 2019-04-11:
Thanks @slurdge your post gave me confidence. I read your other post also, how did you flash without using the adapter shield. Did you use some type of special wire to solder it? Is there any pic u can share how u did it. Also did u use the 6 pin or 8 pin connector from Avr isp mkii?

slurdge, 2019-04-11:
I will edit the wiki with a picture when I have time (probably tomorrow).

dealaddict, 2019-04-11:
Thank you @slurdge  By the way I am having the exact same problem as yours. It seems quite rare haven't seen many such posts.

- Dim light blink upon connection
- Light blinks near nfc device
- USB device connects can hear sound on computer then gives device descriptor error in device manager
- Cant boot into bootloader DFU
- Tried @iceman1001  no light flashing post by renaming those two eep files. It stills no device present
- So I eventually bought MKii but did not realise I needed the additional connector shield.

slurdge, 2019-04-17:
Ok @dealaddict please look at the following wiki page:
[How-to-program-with-AVRISP-mkII](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/How-to-program-with-AVRISP-mkII)
You should be able to salvage your unit. Please not that eep is *eeprom only*. You need hex for flash.

dealaddict, 2019-04-18:
@slurdge  Thanks a bunch bro. That post is going to be very helpful not just to me but for many others. Seems like I would need a breadboard with those jumper cables so I will order one soon and go from there.

closed 2019-04-29 by dealaddict