iceman1001/ChameleonMini-rebooted #92: legic UID emulation

chaytoun911, 2019-03-30:
I got my Chameleon Mini REV E , I'm in deep testing with some of my clients to prove that UID is not secure, the client is running on Legic Advant cards which looks to be @ 7 Bite UID 

is there any way I would be able to emulate Legic 7 Byte over the Mini REV E ?

chaytoun911, 2019-04-01:
Anybody has a feedback on this topic? thanks for your support.

bogiton, 2019-04-01:
As it is, since there is no support for Legic, you can't. Also, aren't those iso15? Currently, there is only support for iso14443a.
You are free to add support for Legic though if you want! It would be more than welcome. :)

chaytoun911, 2019-04-01:
Sorry missed adding, Card type shall be ISO14443 in above scenario so I can do the proper card emulation.

bogiton, 2019-04-01:
You still have to select an existing configuration for you base, like MF Classic or UL. I am not sure that changing the ATQA/SAK would make the reader believe that it has to do with a Legic. It all comes down to the communication between the card/Chameleon and the reader. Since, there is support only for MF Classic and Ultralights, the Chameleon will respond like those to the requests coming from the reader. On the other hand, it is worth trying it out as it is.

iceman1001, 2019-04-01:
There is no built in-support for ISO7816-4 ,  apdu,  nor is the Mifare Desfire commandset. 

If a simple anticollision is needed then use the ATQA/SAK commands to respond with proper values for Desfire.

chaytoun911, 2019-04-02:
Thanks both, I still have an issue with the required firmware.

Can somebody provide me with a link for the latest updated & compiled firmware.

Hex and eep files are ok.

Linux and compilers are not my core.

Highly appreciated

iceman1001, 2019-04-02:
look and behold,   https://github.com/iceman1001/ChameleonMini-rebooted/releases

closed 2019-04-02 by iceman1001

chaytoun911, 2019-04-02:
Thanks iceman , I'll update the status of testing, I'm ready to support based on my RF knowledge to help in developing further more the firmware of the chameleon REVE 

This will be a great tool for security specialist, being part of Security community, I can guarantee you a lot of support if we are able to provide something which security specialist can use on daily basis.