iceman1001/ChameleonMini-rebooted #95: DFU doesn't work

stek29, 2019-04-10:
Device never goes to DFU mode -- I have to flash it with external programmer.

Neither holding button nor sending "UPGRADE" over serial work -- it still boots in normal mode.

I have Chinese RevE rebooted version.

iceman1001, 2019-04-10:
Does it go into DFU mode with the latest source from here?

slurdge, 2019-04-10:
I think you have to send `MYUPGRADE` and not `UPGRADE`

iceman1001, 2019-04-10:
yeah,  I normally use the GUI for chameleon.

dealaddict, 2019-04-10:
@stek29  can you explain how you program with external programmer? I bought a avr isp mkii myself and need some instructions. Thanks

stek29, 2019-04-15:
@iceman1001 I've tried with latest master.
In fact, it did get into DFU mode with original Chinese firmware with button.

(sorry for late reply)

slurdge, 2019-04-17:
@stek29 Is your issue resolved?

stek29, 2019-04-18:
@slurdge No, why do you think it is?

stek29, 2019-04-18:
@dealaddict see https://github.com/iceman1001/ChameleonMini-rebooted/issues/10#issuecomment-481510898

stek29, 2019-04-18:
Can somebody with working DFU post output of `avrdude -D -v -p atxmega32a4 -P usb -c avrispmkII -U fuse0:r:-:i -U fuse1:r:-:i -U fuse2:r:-:i -U fuse4:r:-:i -U fuse5:r:-:i` so I can compare fuse bits?

stek29, 2019-04-18:
Hm, maybe I've erased boot loader long time ago, and there's simply no boot loader to get into.
(related: #11)
I've tried both RevE-atxmega32a4u_104_modified.hex and one from AVR1916 bootloader -- both are invalid for my chip since boot loader starts at 0x8000, while my model has only 0x8000 bytes in `application` memory.

According to Atmel's XMEGA manual, XMEGA A4 can have `16 - 128` KBs of program memory, and looks like mine has 32 -- exactly 0x8000 (according to verbose info from avrdude and just judging by name).

Judging by XMEGA manual, boot `section` goes after `application` and `apptable`, so next logical thing to do was to flash boot loader directly to boot.

So I ended up shifting bootloader file 0x8000 bytes backwards:

```
❯ srec_info RevE-atxmega32a4u_104_modified.hex -intel
Format: Intel Hexadecimal (MCS-86)
Execution Start Address: 00008000
Data:   8000 - 81A5
        81F4 - 8FB5

❯ srec_cat RevE-atxmega32a4u_104_modified.hex -intel -offset -0x8000 -o RevE-atxmega32a4u_104_modified_shifted.hex -intel

❯ srec_info RevE-atxmega32a4u_104_modified_shifted.hex -intel
Format: Intel Hexadecimal (MCS-86)
Execution Start Address: 00000000
Data:   0000 - 01A5
        01F4 - 0FB5
```

and flashing it to `boot`:

```
❯ avrdude -e -vv -p atxmega32a4u -P usb -c avrispmkII -U boot:w:RevE-atxmega32a4u_104_modified_shifted.hex:i
```

And now I can go to DFU both by sending `UPGRADE` command and by holding the button.

closed 2019-04-18 by stek29