iceman1001/ChameleonMini-rebooted #97: Some questions about Chameleon Mini GUI

Ollibolli85, 2019-05-07:
Good evening everybody,

i finally own a „Chameleon Mini RevE“ and almost everything works very well with the newest Firmware/“Chameleon Mini GUI“!

Why it is only possible to set up just „Slot 1“ with „Mifare 4K“-settings on „Chameleon Mini GUI“?

If  i open „Chameleon Mini GUI“ without the „Chameleon Mini“ connected, there are also many more configurations for each slot!?

Many thanks in advance!

bogiton, 2019-05-09:
> Why it is only possible to set up just „Slot 1“ with „Mifare 4K“-settings on „Chameleon Mini GUI“?

Before we knew that there is some extra memory space on the RevE rebooted board, we decided to enable Mifare 4K option only to the first slot to save memory space, in order to add more features to the firmware. At some point, we will make use of the extra memory and enable Mifare 4K option in the other slots too.

> If i open „Chameleon Mini GUI“ without the „Chameleon Mini“ connected, there are many more configurations for each slot!?

When you connect a device, the slots get enumerated by querying the supported features of that device. The GUI supports both the RevE rebooted and the RevG one.

> What Firmwares are also available?

The stock firmware (that came with your board) and the iceman version in this repo.

Ollibolli85, 2019-05-10:
Thank you very much for the fast answers!

Ollibolli85, 2019-05-12:
My Chameleon Mini is just recognized/working by 1 of 2 devices i need it for!

If i use a cloned Mifare-tag on the same devices, everything works fine!

What can be the reason for that?

iceman1001, 2019-05-12:
Details,  details,  without details, no answers.

Ollibolli85, 2019-05-12:
The 2 devices work fine with a cloned Mifare 2K-tag!

If i load the same dumpfile.bin-file to the Chameleon Mini (GUI), the first device recognize the Chameleon Mini as it should be, but the second device don't do/show anything!

"MF_DETECTION" is working with the second device, but i already know the keys, because the cloned Mifare 2K-tag works fine here!

bogiton, 2019-05-12:
What is the device that it isn't working with?
Is it possible to provide traces of the communication between the cloned MF tag and the device, as well as between the Chameleon and the device? For example using a PM3 if you have one?

Ollibolli85, 2019-05-12:
I own a Proxmark3 "Easy"!
Is it possible to sniff the traces between the devices without a Notbook?

bogiton, 2019-05-17:
@Ollibolli85 You could use an Android phone with an app like AndProx. Or even a standalone mode. I'm not familiar with the "Easy" version though.

closed 2019-05-18 by iceman1001