iceman1001/ChameleonMini-rebooted #98: bootloader mode

nikmel420, 2019-05-12:
ok my revE has never went to bootloader mode. i always used the widows gui. only thing i think i use windows for. welli got all files ready on my main computer. but couldnt figure out making the gui. so tranfered all files to windows. the gui wont open all the way no mater what ive tryed the botom doesnt show. i spent a day changing resolutions. pixel size ect. cant get botom to show. so i cant get to bootloader mode. is the a way to short (jump) to bootloader mode? i cant get it to work on windows cmd line. any idea be helpfull. or if you have a copy of an old like 1st or 2nd version gui. (and im very flatered you named the main directory after me for my stupidity im immortal now)

bogiton, 2019-05-17:
Have you read this: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Getting-started#get-ready-to-flash-bootloader-mode ?
You can find the stock software (including old GUI) here: https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Software/Win32
The new GUI can be found here: https://github.com/iceman1001/ChameleonMini-rebootedGUI

shinhub, 2019-08-12:
Could we close this @iceman1001 (doc provided, and issue seems more with GUI size than with actual firmware)? Who can actually close issues there?

closed 2019-08-12 by iceman1001