iceman1001/ChameleonMini-rebooted #100: Android MCT won't read emulated cards

nicoschtein, 2019-06-10:
MCT reads UID fine, but fails with either "card disconnected" error along the way while building key map, or "key not found" error, which is not correct since the key is present in the selected keyfile.

Tested with multiple dumps and phones, and latest FW and GUI versions.

Is anyone able to read any card from the chameleon with MCT android app?

Skrunchamunch, 2019-06-11:
I find it difficult to read the card with MCT.  It seems very timing sensitive.  I keep hitting the "Start Mapping and Read Tag" button and once in a while, I get lucky.  Also, if the Chameleon moves even the slightest bit, the reading fails.  Could be an antenna issue.

With the ACR122U reader, I never seem to have an issue.

McEloff, 2019-07-10:
There is a trace log between Chameleon RevE (tag) on @iceman1001 firmware and Android smartphone (reader) :

[log.txt](https://github.com/iceman1001/ChameleonMini-rebooted/files/3376761/log.txt)

As you might seen, Chameleon replies "AT" of succeeded authentication, but Android not accepts this "AT" and sends not encrypted HALT command (50  00  57  cd) in encrypted mode. 

Interesting fact, the Chinese firmware for Chameleon RevE (that based on @iceman1001 firmware) does not have this effect, look at trace log:

[log_china.txt](https://github.com/iceman1001/ChameleonMini-rebooted/files/3376815/log_china.txt)

McEloff, 2019-07-11:
I think, there is a slow calculation of AT-response. This @iceman1001 firmware performs about 15000 ticks to decrypt ar-nr request and calculate at-response. Stock Chinese firmware performs about 10000 ticks (up to 30% better). I solve this issue by implementing into @iceman1001 firmware 'Crypto1.c' from official repository (see attach):

[crypto1.zip](https://github.com/iceman1001/ChameleonMini-rebooted/files/3381378/crypto1.zip)

After that Android smartphone does work corrected. Firmware performs AT-response about 9000 ticks, look at log:

[android_log.txt](https://github.com/iceman1001/ChameleonMini-rebooted/files/3381401/android_log.txt)

bogiton, 2019-07-11:
Very good @McEloff! I want to try this! Can you do a PR with the changed file?

Skrunchamunch, 2019-07-11:
Wow, nice work!!!  I'm going to try the patch.

McEloff, 2019-07-11:
@bogiton You can extract files from attached crypto1.zip and exclude original Crypto1_asm.* from compilation to check locally. I'm not commit my changes to remote repository yet. For full Android corrected works firmware must contains other method to nested authentication. Now I fixing this

slurdge, 2019-07-11:
Very nice 👍

slurdge, 2019-07-11:
@McEloff I tried your patch but it didn't change much in my case. I have UID working well but not key mapping with MCT.

McEloff, 2019-07-11:
@slurdge Under firmware build from my repo MCT succeed maps keys and read tag

slurdge, 2019-07-11:
I build from your repository. I succeeded to map keys and read tag but it was still quite difficult. Anyway that's good progress! Congratulations!

iceman1001, 2019-07-12:
I had some respone to the PR,   I should have writting it here instead.
three kinds of crypto1 implementations.
RevG . latest
RevE rebooted old
RevE rebooted new (iskuri)

RevG implementation relies on the large mcu.  Not suitable from reve rebooted.
Old revE rebooted version,  was apparently too slow for simulations.
New revE rebooted version,  should have been faster...

Now I feel we need to test all of them and really document speed, memory usage,  which version is the most suitable for this project.

iceman1001, 2019-07-30:
@slurdge   May try now that fix for full  byte anticollision is merged?

slurdge, 2019-07-30:
I tried both master and the @McEloff branch. On master, situation is more or less the same, I can't reliably read a tag with my phone. On McEloff branch (with crypto1 changes), I can now read tag (at least 3 times in a row if correctly placed).

bogiton, 2019-08-03:
Same as @slurdge regarding MCT. I even tested the stock firmware. None of them was accepted by the real reader though. But since we are moving forward I would go with @McEloff implementation for the time being.

iceman1001, 2019-08-14:
So,   tests,  tests, please.
Does it work on Android reading the RevE rebooted with latest code as of today?
report back so I can close this one.

iceman1001, 2019-08-18:
@McEloff   Would you mind testing latest source and see if there is any more Mifare Classic issues with crypto,  readers,  1k , 4k , 4b , 7B ?

shinhub, 2019-09-13:
@iceman1001 great, exciting. Should we close this MCT issue then?

closed 2019-09-13 by iceman1001