iceman1001/ChameleonMini-rebooted #104: Sniffed keys are not usable with mfoc

Wolfgangus22, 2019-07-22:
I used the GUI from Iceman
https://i.imgur.com/dwxA4cd.jpg
set the mode to MF_DETECTION, clicked mfkey32, and this was the outcome

> [+] Success, found Chameleon Mini device on 'COM3' with Firmware RevE rebooted installed
> [Tag slot 1]
> [S0 / B1] KeyB [f0715dfb9afc]
> [Tag slot 2]
> [S0 / B1] KeyB [9b028235016e]
> [Tag slot 3]
> [S0 / B1] KeyB [299bdf4d726d]
> [Tag slot 4]
> mfkey32 attack failed, no keys found
> [Tag slot 5]
> [S0 / B1] KeyB [78aaa11efade]
> [Tag slot 6]
> [S0 / B1] KeyB [78aaa11efade]
> [Tag slot 7]
> mfkey32 attack failed, no keys found
> [Tag slot 8]
> [S0 / B1] KeyB [78aaa11efade]

I created  `keys.txt` with the keys found and fired my ACR122
```
root@kali:~# mfoc -f keys.txt -O mycard.mfd
The custom key 0xf0715dfb9afc has been added to the default keys
The custom key 0x9b028235016e has been added to the default keys
The custom key 0x299bdf4d726d has been added to the default keys
The custom key 0x78aaa11efade has been added to the default keys
Found Mifare Classic 1k tag
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
* UID size: single
* bit frame anticollision supported
       UID (NFCID1): 0b  ef  55  9d  
      SAK (SEL_RES): 08  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Classic 1K
* MIFARE Plus (4 Byte UID or 4 Byte RID) 2K, Security level 1
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:

Try to authenticate to all sectors with default keys...
Symbols: '.' no key found, '/' A key found, '\' B key found, 'x' both keys found
[Key: f0715dfb9afc] -> [................]
[Key: 9b028235016e] -> [................]
[Key: 299bdf4d726d] -> [................]
[Key: 78aaa11efade] -> [................]
[Key: ffffffffffff] -> [................]
[Key: a0a1a2a3a4a5] -> [................]
[Key: d3f7d3f7d3f7] -> [................]
[Key: 000000000000] -> [................]
[Key: b0b1b2b3b4b5] -> [................]
[Key: 4d3a99c351dd] -> [................]
[Key: 1a982c7e459a] -> [................]
[Key: aabbccddeeff] -> [................]
[Key: 714c5c886e97] -> [................]
[Key: 587ee5f9350f] -> [................]
[Key: a0478cc39091] -> [................]
[Key: 533cb6c723f6] -> [................]
[Key: 8fd0a4f256e9] -> [................]

Sector 00 - Unknown Key A               Unknown Key B
Sector 01 - Unknown Key A               Unknown Key B
Sector 02 - Unknown Key A               Unknown Key B
Sector 03 - Unknown Key A               Unknown Key B
Sector 04 - Unknown Key A               Unknown Key B
Sector 05 - Unknown Key A               Unknown Key B
Sector 06 - Unknown Key A               Unknown Key B
Sector 07 - Unknown Key A               Unknown Key B
Sector 08 - Unknown Key A               Unknown Key B
Sector 09 - Unknown Key A               Unknown Key B
Sector 10 - Unknown Key A               Unknown Key B
Sector 11 - Unknown Key A               Unknown Key B
Sector 12 - Unknown Key A               Unknown Key B
Sector 13 - Unknown Key A               Unknown Key B
Sector 14 - Unknown Key A               Unknown Key B
Sector 15 - Unknown Key A               Unknown Key B
mfoc: ERROR: 

No sector encrypted with the default key has been found, exiting..
```
I'm not sure what I'm doing wrong.

bogiton, 2019-07-28:
I'm not sure I understand what I'm seeing. You have set the MF_DETECTION option to all slots and run mfkey32 to each and every one of them? The UID was the same in every slot? UID should be a valid one for the reader you are testing against in order to get a usable key.

iceman1001, 2019-07-28:
Dunno but  the image shows a UID of  `BC59FC0F  `and the card you are trying the keys against has a UID of  `0BEF559D`

The keys could be diversified by UID,  so one set o f keys doesn't work on other cards from same system

Wolfgangus22, 2019-07-29:
The image was just illustrative. I did replace the UID to `0BEF559D` on all slots before running mfkey32.

iceman1001, 2019-07-29:
Exceptional difficult to help someone who doesn't post actual data...

bogiton, 2019-07-29:
Still I don't get why you set the same UID and configuration (MF_DETECTION) on every slot. One should suffice.

iceman1001, 2019-07-29:
Never mind that part,   even if OP used all 8 slots it should work...
One is of course enough but thats a different story.

Wolfgangus22, 2019-07-30:
> Still I don't get why you set the same UID and configuration (MF_DETECTION) on every slot. One should suffice.

I've been trying to test all the slots to get a reliable key. I've tried 3 cards from the same system and I couldn't find a working key for mfoc, so I tried to test all the slots thinking it was a slot problem.

> Exceptional difficult to help someone who doesn't post actual data...

I'm unsure of what other data I should provide. I have 2 other cards but I'm encountering the same problem with them too. In my OP I just tried to recreate the process.

iceman1001, 2019-07-30:
Lets see if the fix for 4k  might have solved your issue @Wolfgangus22

iceman1001, 2019-08-12:
And is there a way for you to test the keys with a proxmark or  acr122u /libnfc ?

iceman1001, 2019-08-14:
@Wolfgangus22   try using it again with latest code from today.

iceman1001, 2019-08-17:
time to close?

closed 2019-09-14 by iceman1001