iceman1001/ChameleonMini-rebooted #116: [Question] Knowing firmware version?

c4nh0t0, 2019-08-16:
Hi.
Sorry for this post because actually isn't a issue but a doubt/question.
Recently I've bought a Chameleon Mini RevE from Hacker Warehouse. The device works fine but I wanna know how can I find what firmware version is running on my device because I wanna upgrade the firmware with **iceman1001 github latest update**.

Thanks in advance.

iceman1001, 2019-08-16:
I believe that all sold Chameleon Mini RevE rebooted comes with the same factory version of the firmware.  Its the same as the one we have here in the repo for recovering if things goes bad. 
It has a firmware string,  we renamed the versioning according to what we released from this repo.
@bogiton   which is the firmare string for the factory fw?

shinhub, 2019-08-16:
@c4nh0t0 the best you can find is in the [GUI](https://github.com/iceman1001/ChameleonMini-rebootedGUI) (please check Wiki page about tools), at "Settings" tab, or with "VERSION?" ("VERSIONMY?" if brand new) within a terminal connection to the Chameleon.

Original firmwares for RevE with LEDs appeared as `Chameleon-new-1.0`. Versions from this repo also appeared like this, till February 2019. It now appears as `Chameleon Mini Rebooted X.Y (Iceman)`. X.Y may change in the future, but is "1.3" for now.

There is for now no better way of knowing, except of noticing some user-oriented changes like commands (which used to end with "MY" for older firmwares), and hashing the firmware files (.eep, .hex) to compare with this repo.

iceman1001, 2019-08-17:
That MY-extension was a dick move.  But hey,  locking down open source....

shinhub, 2019-08-17:
Yes, plus they could have at least ensured the "MY" was not noticeable to the user, by appending it automatically in all the command strings received. Because it doesn't make any sense at all. Shouldn't we add ICEMAN as a suffix to all commands now? 😀 

@c4nh0t0 does this solve this issue? If so could you close it?

iceman1001, 2019-08-17:
You think I should make a dick move?   Lets rename it SHINHUB ..  with adding it dynamically,  but not always.

shinhub, 2019-08-17:
@iceman1001 if only there were not already things to fix and clean... I totally see a -DMOD_DICKMOVE compiler directive. Let's settle this to add "DICKMOVE" suffix, and then randomly change to "SHINHUB", "ICEMAN", or "MYMY". Also the firmware name change to another animal, and bytes in memory get swapped.

iceman1001, 2019-08-17:
Hahaha!   I love it!

closed 2019-08-17 by c4nh0t0

iceman1001, 2019-08-17:
So I guess @c4nh0t0  got his version command to work? 
If you use the GUI under the settings tab,  you see it.