iceman1001/ChameleonMini-rebooted #118: [Question] "Reader Attack" Timing

c4nh0t0, 2019-08-17:
Hi.

Do you guys know if Chameleon when used in mode **MF_DETECTION** when approached near to the target reader, the red led blinks... what actually it's doing behind the scenes?
I mean, I'm doing some tests with Chameleon and reader attack with different NFC readers spread in the city. Till now I didn't manage to recover any key.

I didn't find the amount time holding Chameleon near to the reader. It makes any difference holding it for 10 seconds or 10 minutes?
My final question is: it makes any difference let Chameleon near to the target reader more and more minutes? It will make any difference when using **mfkey32** or **fast/exact recon**?

Thanks in advance.

maxieds, 2019-08-17:
Have you tried reading the firmware source in C? It's available if you ever
have questions or need customizations.

iceman1001, 2019-08-17:
if you read some of the older Mifare Classic attack papers,  you will find reference to a 32bits recovery and 64bits recover attack.

c4nh0t0, 2019-08-17:
Since I recently acquire one, the purpose of my questions is helping Chameleon project documenting functionalities and discrepancies using the hardware in different scenarios.

iceman1001, 2019-08-17:
Good initiative

closed 2019-08-17 by c4nh0t0