iceman1001/ChameleonMini-rebooted #127: Flashing problem on Windows with flash.bat

uspilot, 2019-08-18:
Latest version of Windows flash.bat not working on directories with space character on their name.

Several lines have to be updated with **""** enclosing filename to fix it:
```

line 36:    move /y "%dirpath%%oldname%.eep" "%dirpath%%name%.eep"
line 46:    move /y "%dirpath%%oldname%.hex" "%dirpath%%name%.hex"
line 59:    del "%dirpath%eeprom.bin"
line 64:    move "%dirpath%myfile.bin" "%dirpath%myfilee.bin" >nul
line 71:    del "%dirpath%flash.bin"
```

shinhub, 2019-08-18:
@upsilot, thanks for the tip, this has actually been crafted with absolutely no BAT knowledge. Can you write a PR? If not I would probably later in the week.

shinhub, 2019-08-19:
@uspilot thanks a lot for your help.

closed 2019-08-19 by shinhub