iceman1001/ChameleonMini-rebooted #130: Supported Cards and Codecs Wiki page

Mera-balou, 2019-08-19:
Is it possible to add a page dedicated to supported cards and codecs on the wiki?
ChameleonMini RevG wiki has already in place [this page](https://github.com/emsec/ChameleonMini/wiki/Supported-Cards-and--Codecs) that I find very useful.
It also makes it easy to advertise next features roadmap.

BTW, do you plan to support the same cards as the RevG (eg. desfire, ISO 15693...) or there are some hardware limitations on ChameleonMini-rebooted?

shinhub, 2019-08-19:
Hi,
This existing page should answer your question: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Configurations.
The main repo page also links to the page you mention.

There is no clear roadmap of what future standards that would be supported, as we are still working on an early-stage development and fixing bugs first.

iceman1001, 2019-08-20:
@shinhub   Maybe time to add the 4b - 7b UID support for classic in that wiki page.  
That list shows fw from manufacturer support list.   Not what this repo fw does.

iceman1001, 2019-08-20:
@Mera-balou    No,  we don't have a roadmap for this repo.  Last months stuff has been bugfixes.  Nothing new is actively being developed.   Feel free to contribute!

Mera-balou, 2019-08-20:
You've right, i've to engage in the project. But I don't know how to program as well as I want.
I choose to become a Patreon as an alternative. Thx for your work iceman.
PS: Desfire support will be great...

closed 2019-08-20 by Mera-balou