iceman1001/ChameleonMini-rebooted #137: The ATQA doesn't stick for the card dump in my slot

heyquentin, 2019-08-23:
Hi, I've got a problem with changing my ATQA. Here's what I'm working with:

Chameleon Mini RevE Rebooted
--> VERSION?
101:OK WITH TEXT
ChameleonMini-rebooted v1.3 (Iceman: 7420671)

GUI 1.2.0.19

Slot 1 Mode: MF_CLASSIC_1K
Card dump loaded in slot 1

I'm trying to emulate a Mifare 1k card that has ATQA 0004. When I change the ATQA on my emulated card from 0044 to 0004 it says it changes but then if I set another slot active and come back to slot 1 the changes revert. Why is this? Am I doing something wrong?

[image]

shinhub, 2019-08-24:
Hi,
We cannot see that your slot is actually configured as "MF_CLASSIC_1K" here, and some informations are missing on your dump and context:
- Could you reset a slot, and show us all of the steps from slot configuration to final ATQA reading?
- Are you sure the card dump you uploaded in Mifare Classic 1K?
- Is the UID 4 or 7bits long in your card dump/your configuration?
- Did you change anything in your Makefile before compiling?

The slot configuration will set a default ATQA, and if you upload a dump, ATQA will also be reset according to the dump. You should not have to set any ATQA by yourself in your scenario ("trying to emulate Mifare 1k") though.

heyquentin, 2019-08-24:
Hello, thanks for the reply. 

> * Could you reset a slot, and show us all of the steps from slot configuration to final ATQA reading?

I certainly can. I thought a screen recording would be easier than attempting to explain via text and images. You may view the recording here: https://youtu.be/j50C4COT8Z8

In the video you can see first a check to make sure the Chameleon is connected as well as a firmware check. I then select slot 1 and clear its' configuration. Through the serial connection I make sure the slot is selected and check the ATQA. Once it's verified as closed, I open the slot and set it appropriately with the mode and UID of my card. Then I check the slot via serial and verify the ATQA. Note at this point no dump has been loaded. I try changing the ATQA and it appears successful however when I switch slots and then switch back we can see the ATQA reverts to 0044. I then load up my dump and attempt the process again. We see that the same behaviour persists regardless of if a dump is loaded to the slot or not.

> * Are you sure the card dump you uploaded is of Mifare Classic 1K type?

Yes I believe it is. Here is a screen recording of the card being checked with my Proxmark: https://youtu.be/IVqEt1c7_IM

> * Is the UID 4 or 7bits long in your card dump/your configuration?

My understanding is that it's 4 but please do correct me if I'm wrong
[image]

> * Did you change anything in your Makefile before compiling?

I did not. I downloaded a compiled firmware by following the instructions on [this page](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Getting-started#get-or-compile-a-firmware). I used the latest manufacturer compiled firmware is from the [lab401 link.](https://lab401.com/products/chameleon-mini/firmware/latest.zip)

> The slot configuration will set a default ATQA, and if you upload a dump, ATQA will also be reset according to the dump. You should not have to set any ATQA by yourself in your scenario ("trying to emulate Mifare 1k") though.

If I understand correctly, my dump is responsible for setting the ATQA? I thought that might be the case so before posting this question I attempted to locate the part of my dump that was responsible for the ATQA but I was not able to do so. Could you please point me in the right direction?

shinhub, 2019-08-24:
There was actually a bug in MifareClassic emulation, where 4B and 7B ATQA codes were inverted at slot initialisation. This is being corrected, please try with a newly compiled firmware from this repo (we cannot actually support the firmwares you get from a third party).

However, in the way you use the GUI, the serial commands and graphical configuration might be competing with each other.
You are supposed to chose a card type in slot, and upload a dump if you will. The ATQA is determined by the card type you choose (please search for MifareClassic datasheets references on the Internet if you need to understand this), and you should not have to set it by hand.
Your card is a 4B UID, but due to the bug, it was reverted to a 7B UID type ATQA each time you set something for the Mifare slot.

@iceman1001 is this actually OK that Lab401 distributes compiled firmwares from this repo with products they sell, without even putting the KAOS/emsec license anywhere, or telling it comes from this open-source repo?

heyquentin, 2019-08-24:
Thank you for the explanation. I will compile from this repo, flash and report back my findings hopefully either today or tomorrow.

Thank you for correcting my usage on the GUI. I appreciate it.

shinhub, 2019-08-24:
No problem, you did not do wrong - it is just that this firmware might be quite unstable at it is still in development. Thanks for your help identifying a bug in any case.

heyquentin, 2019-08-25:
Everything looks to be working as intended after the fix. The dump isn't working on my reader but that's not your problem :) Thanks for the help!

iceman1001, 2019-08-25:
that is their problem,  not mine.  KAOS bros informed me that this repo didnt follow their license, so I modified it and since then I haven't heard anything more.   ie,  its KAOS bros who should inform and complain to LAB401.  Not us.

closed 2019-08-25 by iceman1001