iceman1001/ChameleonMini-rebooted #139: [REQ] Implementing support for MF_DETECTION on Rev G

lavanoid, 2019-08-24:
### Description

I'm currently attempting to bodge your MF_DETECTION feature into the RevG firmware. I honestly have no idea what I'm doing wrong and I'm by no means an expert when it comes to C and writing code for hardware.

Here is the commit I have made so far, on a testing branch: https://github.com/lavanoid/ChameleonMini/commit/0ad00933b5aeca77cb0722e145d630c656dbe768 
### Interest

I just recently bought a RevG, as it is supposedly newer than RevE. So I was under the impression that the MF_DETECTION feature is available in the RevG. Now I understand that this feature is only available in the rebooted firmware for the RevE, and is a seperate project.
### Possible implementation

Currently trying to transplant the current code base of RevERebooted to RevG official: https://github.com/lavanoid/ChameleonMini/commit/0ad00933b5aeca77cb0722e145d630c656dbe768

shinhub, 2019-08-24:
Dear @avanoid,
Thanks for sharing your interest.
As specified in various places on this repo, including main page and Wiki, this repo is dedicated to RevE Rebooted only.
You will want to share your thought on RevG issues pages: 
Good luck with your attempt at implementing this, hoping this code repo will be a useful source.

https://github.com/emsec/ChameleonMini/issues

closed 2019-08-24 by shinhub

lavanoid, 2019-08-24:
Okeydoke, thank you.

I was hoping if someone could at least check my commits and point out what could be wrong? I know this repo is dedicated to the RevE Rebooted but some insight by someone who is more informed would be appreciated, so I can at least make my own fork of the RevG firmware with this feature. Don't necessarily need to adapt this repo to support both devices.