iceman1001/ChameleonMini-rebooted #148: DOWNLOAD from closed slot return 400k dump

uspilot, 2019-09-14:
Download command on closed slot return dump of all memory or something else with size of about 400k.

## Environment

|Item|Your information|
|---|---|
|**Harware**|RevE Rebooted
|**Firmware**|ChameleonMini-rebooted v1.3 (Iceman: b0ccfb0)|
|**GUI**|https://github.com/iceman1001/ChameleonMini-rebootedGUI|
|**Slot number**|4|
|**Slot configuration**|CLOSED|
|**Dump source**|N/A|
|**Reader**|No case|
|**Flashing environment**|Compiled HEX/EEP/EXE|
|**Flashing method**|USB / flash.bat|
|**Flash memory space**|512K ? / NO SPI_INFO command|
|**Makefile configuration**|N/A|

## Bug description
DOWNLOAD command from CLOSED slot using SERIAL terminal in GUI return dump of about 400K. At the same time <Download> button save file with 16 bytes (which equals to slot size), but reading time is the same as in terminal mode. 
### Expected function and references
Download dump according to slot size (16 bytes) or none (slot is CLOSED).
### Bug
```
--> SETTING?
101:OK WITH TEXT
NO.4

--> CONFIG?
101:OK WITH TEXT
CLOSED

--> MEMSIZE?
101:OK WITH TEXT
16

--> DOWNLOAD
110:WAITING FOR XMODEM

ff ff ff ff 00 ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
...
```

### Steps to Reproduce
1. Set slot to CLOSED state.
2. Make them active.
3. Send `DOWNLOAD` command in terminal

## Resolution paths
### Ideas
N/A
### Possible Implementation

iceman1001, 2019-09-14:
@shinhub   some more testing?

shinhub, 2019-09-14:
@iceman1001 well... downloading a closed slot, what would you expect? : ) At least it is supposed to download that quicker than before :p 
Seriously, I did not manage to understand how XModem Download is supposed to be limited in downloaded bytes somehow, so it just downloads all available memory for the slot I guess, until memory layers stop by application memory space boundary.
I knew I had to get around this, but it should not happen for other slots (it did not for me), and I still do not know how to limit XModem download size...

shinhub, 2019-09-14:
Gosh the PR auto-closed it... well @uspilot , please try with latest and tell us.

uspilot, 2019-09-15:
fixed. 
thanks @shinhub

closed 2019-09-15 by uspilot

shinhub, 2019-09-15:
@uspilot thanks to you spotting this and describing it correctly. I would not even have tried to download a closed one : )

PR #149 (2019-09-14): Memory polish
Move EEPROM system init to memory rewrite, fix #148 by better limiting XModem upload/download R/W sizes, allow flash erase for spaces that are less than a page, and fix "CLOSED" settings standard data returns (I noticed they were generating random RAM reads).