iceman1001/ChameleonMini-rebooted #154: [BUG] MF_CLASSIC_1K_7B Setting UID fails if set on multiple slots

pa4hjh, 2019-09-16:
## Environment

|Item|Your information|
|---|---|
|**Harware**|RevE Rebooted|
|**Firmware**|ChameleonMini-rebooted v1.3 (Iceman: 19af655)|
|**GUI**|Chameleon Mini Gui - v1.2.19 - Iceman Edition|
|**Slot number**|1-2| 
|**Slot configuration**|MF_CLASSIC_1K_7B||
|**Dump source**|N/A||
|**Reader**|Lab401||
|**Flashing environment**|N/A||
|**Flashing method**|N/A||
|**Flash memory space**|N/A||
|**Makefile configuration**|N/A||

## Bug description
### Expected function and references
Setup multiple MF_CLASSIC_1K_7B slots with UID's
### Bug
Based on the latest github changes I saw that some changes have been made to improved 7b support. Using the icemangui and the latest FW dump I tried the updated feature.
If I program "Slot 1" with a 1k_7b key everything works fine. As soon as I program the "Slot 2" with a 1k_7b key, the key key which is written (or read back) is different from what I've entered.
If I clear the 2nd slot and try it again it will only read back FFFFFFFF no matter what I use as a key.
### Steps to Reproduce
Activate Slot 1
Select Mode: MF_CLASSIC_1K_7B
Write 7 byte UID in UID field
Button: unchanged (closed)
BtnLong: unchanged (switchcard)
Size: 1024
Click Apply
UID reads back correctly.

Deselect Slot 1
Select Slot 2
Select Mode: MF_CLASSIC_1K_7B
Write 7  byte UID in UID field
Button: unchanged (closed)
BtnLong: unchanged (switchcard)
Size: 1024
Click Apply
Unkown UID (seems random) is read back after write completes.

Clear Slot 2
Select Mode: MF_CLASSIC_1K_7B
Write 8 byte UID in UID field
Button: unchanged (closed)
BtnLong: unchanged (switchcard)
Size: 1024
Click Apply
UID "FFFFFFFFFFFFFF" is read back from the device.

## Resolution paths
### Ideas
N/A
### Possible Implementation
N/A

N.B.
The previous versions (6 months and older) I was using had similar issues after manually activating the 7b mode in the config file. I worked around this at the time by statically compiling the UID's in the firmware which worked but is not

shinhub, 2019-09-16:
This is most probably be due to the fact that currently published rebootedGUI does not correctly get/set UIDs on some cases. This may have been fixed already, but is not published yet (@iceman1001).
Please use a full terminal interaction (i.e. `minicom`) with this repo firmware to confirm you can reproduce this issue on non-GUI scenario. Start any test by issuing a `CLEARALL` command.

pa4hjh, 2019-09-17:
@shinhub Thank you for your clarification, manually setting the UID via a terminal works as expected.
I've verified the keys by reading them back via the terminal, the gui and an external RFID reader.
@iceman1001 this seems to be related to the GUI.

iceman1001, 2019-09-17:
@shinhub  I have published v1.2.1.0    And this is wrong repo to comment GUI.

closed 2019-09-17 by shinhub