iceman1001/ChameleonMini-rebooted #158: draft of V1.4

iceman1001, 2019-09-22:
A draft for v1.4 release of this firmware has been made.
@shinhub @bogiton @slurdge 
Feel free to update the text to cover all changes.

iceman1001, 2019-09-22:
Those who are collaborators find it under code/releases  -- edit

shinhub, 2019-09-22:
Done. Think I covered most of it (by going through closed PRs list...). Did not mention anything about boot loader, because it is not directly related to firmware, but a lot have been done there as well (we have a working source code for ChameleonMini RevE Rebooted, provided you can flash with AVRISPmkII).

slurdge, 2019-09-22:
I must be blind, can't find where I should edit ❔

iceman1001, 2019-09-22:
you weren't a collaborator.  You got invite.

slurdge, 2019-09-22:
Nice 😄 
BTW super impressive work @shinhub ! I have to flash again my Mini

shinhub, 2019-09-22:
Yeah I messed a lot with this firmware for a start,... I hope someone else than 1 or 2 recent issues openers and me will test that all before it is released in the wild :p

iceman1001, 2019-09-23:
Jean-Michel the grey.   :)  And its still a draft who knows what happens

bogiton, 2019-09-27:
Good job @shinhub! I believe you are the only one who can validate the changelog of this release. You did almost everything by yourself.

shinhub, 2019-09-28:
Thanks! Well that's OK to me, but I think all of Mifare Ultralight work is from @McEloff, so... let him check as well. As for validating release, I would say let it be a few more days for users to identify bugs, and let release somewhere early October if everything fine till then.

iceman1001, 2019-09-28:
We have no stress to make this release.  Let it sip in for a month.

iceman1001, 2021-01-21:
No stress and over a year later,   with other issues instead.  
Somehow I get the feeling we ain't going to do a new release

closed 2021-07-02 by iceman1001