iceman1001/ChameleonMini-rebooted #160: [REQ] ATQA and SAK should be R/W from/to application memory

shinhub, 2019-09-22:
### Description
ATQA and SAK values should be written to application memory, and read from it.

### Interest
This would allow application to take into account any "custom" value that might be set in a dump, and/or to allow a reader getting ATQA and SAK values from reading S0/B0, as well as ensure consistent behaviour when using `ATQA=` and `SAK=` commands.
This is nearly a bug, in fact : )

### Possible implementation
Working on it, in PR #159 .

iceman1001, 2019-09-22:
Not sure I understand what you are considering a bug and with that I am not sure if you are understanding what the optional SAK/ATQA is used for.

shinhub, 2019-09-22:
Well that was not clear to me that "optional" SAK/ATQA setting should be kept separate from what can be read from Block0. If everyone is expecting that then, can be closed.

iceman1001, 2019-09-22:
The ongoing battle between PACS companies readers and cloners,   is that block0 is only informative. The UID/SAQ/ATQA should come regardless what block0 says.   So they pushed a block0 with a SAK 0x80 and in anti collision the reader expected a different SAK.   So all magic cards that uses block0 for this purpose was identified as a clone and promptly locked out of system.
French VIGIK system.

So in order to use chameleon on such systems,  we need a optional way to override UID/SAQ/ATQA...   where we introduced the new commands.  Giving the chameloen for once a sporting chance against those types of systems.

shinhub, 2019-09-22:
Fair enough! Well that's hurting my hunger for consistency, but even Mifare data sheets seem to say we should at least "do not rely" on these SAK/ATQA from Block0.

closed 2019-09-22 by shinhub

iceman1001, 2019-09-23:
making a good research tool doesn't mean just that it is good at following the given standard.
It should also bring options, to see what is more possible.