iceman1001/ChameleonMini-rebooted #161: [REQ] Mifare Classic S0/B0 should be readable without any authentication

shinhub, 2019-09-22:
### Description
The Block 0 in Sector 0 of any Mifare Classic emulated memory should be readable without any authentication, as expected by some readers, and implied by specs.

### Interest
This may fix some inconsistency with some readers, like MCT notably, and may be expected by some readers anyway.

### Possible implementation
Not sure how to "skip" the authenticating phase, as not sure how the readers would expect the tag to react. I think if S0/B0 is requested in "READY"-like state, the application should directly jump to "AUTHED_IDLE" and set a bool to register the direct jump, to only allow READ operation on S0/B0 then.

iceman1001, 2019-09-22:
S0/B0  accessrights should follow what the ST is set for S0.  If  ACR is set non read,  than block0 shouldn't be read.

UID / SAQ /ATAQ is used in anticollision phase,  but that doesn't implose that S0/B0 should always be readable what called with a read command.

shinhub, 2019-09-22:
OK, that was not so clear to me as MCT apparently tries to read it and fails to detect a valid tag if it can't.
I was mislead by this [slideshow](https://www.slideshare.net/GlenCifuentesToro/mifare-cards) from Gemalto (page 13), which says "_can be read without authenticating with sector 0_", but a remark (page 17) of [MF1S50YYX_V1](https://www.nxp.com/docs/en/data-sheet/MF1S50YYX_V1.pdf) says "_The Read from Block 0 in Sequence 2 does not require a prior authentication to Sector 0 and is transmitted in plain data. For all other sequences, the readout from Block 0 in Sector 0 is encrypted and requires an authentication to that sector_".

iceman1001, 2019-09-23:
Let's see,  this issue wasn't the same as sak/atqa.  
That note is interesting,  1) its from Ev1 datasheet,  2) sequence 2 reference. 
What does older models datasheet say?
and which is sequence 1/2 ?

shinhub, 2019-09-28:
- Sequence 1: ISO/IEC 14443-3 compliant anti-collision and selection using the cascade level 1 followed by the cascade level 2 SEL command
- Sequence 2: using cascade level 1 anti-collision and selection procedure followed by a Read command from block 0
- Sequence 3: ISO/IEC 14443-3 compliant anti-collision and selection using the cascade level 1 SEL command

From same page, cannot find any other reference. This for instance cannot be read in MF1S503x.

PR #165 (2019-11-05): Mifare Clean
Factored Mifare code more (mostly anticol, authentication and DETECTION code), ensured all MF1S50YYX_V1 anticol and HALT methods are implemented (this notably fixes #161), and fixed DETECTION bugs.
This results in a smaller code (3% on my side, with AVR-GCC 8 and 9), more reactive Mifare simulation (notably on Android MFC), and DETECTION working (again ?).
To be tested with recents serial and DETECTION fixes in GUI (or DETECTION may give expected results in GUI).
closed 2019-11-05 by iceman1001