iceman1001/ChameleonMini-rebooted #164: Flashing Linux

dose13seitz, 2019-10-05:
I am facing exactly the same problem as described in #93:

> Checking memory from 0x0 to 0x7FFF...  Not blank at 0x1.
> Erasing flash...  Success
> 0%                            100%  Programming 0x20 bytes...
> [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
> 0%                            100%  Reading 0x400 bytes...
> [ X  ERROR
> Memory read error, use debug for more info.
> FAIL

When trying to flash again with the following command:

> sudo dfu-programmer atxmega32a4u flash --eeprom --force --suppress-validation home/dose/ChameleonMini-rebooted/Firmware/ChameleonMini/ChameleonMini.eep.scramble

I receive an interesting error message:

> Error opening home/dose/ChameleonMini-rebooted/Firmware/ChameleonMini/ChameleonMini.eep.scramble
> See --debug=51 or greater for more information.

Any advise on how to proceed? 
Thank's in advance!

bogiton, 2019-10-06:
Did you generate the scrambled files using the crypt_operations.py script, as described here: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing-Linux-(Unix)#flashing-the-reve-rebooted-device ?

dose13seitz, 2019-10-06:
Yes I followed exactly that procedure. One one thing I noticed is that dmesg | grep -i usb gives me the following output: 
> [ 9378.226915] usb 2-1.2: new full-speed USB device number 4 using ehci-pci
> [ 9378.337304] usb 2-1.2: New USB device found, idVendor=03eb, idProduct=2fe4
> [ 9378.337311] usb 2-1.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0

I would have expected an idProduct of 2fe4 similar to the configuration of the udev rules.

> # Rule for ChameleonMini RFID Research tool (RevE)
> SUBSYSTEMS=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2044", GROUP="users", MODE="0666", SYMLINK+="chameleonE", ENV{ID_MM_DEVICE_IGNORE}="1"

bogiton, 2019-10-06:
The product ID should have been 2fde when in DFU mode. I guess you can always change the corresponding udev line for it. Have you tried flashing the device under Windows using the BOOT_LOADER_EXE?

securechicken, 2020-02-23:
Closed, no feedback.

closed 2020-02-23 by securechicken