iceman1001/ChameleonMini-rebooted #169: [REQ] Allow writings on OTP and locked pages when simulating mfu tags

BreakSecurity, 2019-11-12:
### Description

Would be nice to have the possibility to write on OTP and locked pages when simulating mfu tags.
### Interest

Many times you do not have the possibility to uploads dump with a PC, especially on field testing. So the possibility to modify those pages is really useful

### Possible implementation

Add a MODE called maybe 'OTP&LOCK_WRT' that let you write on those page when button is pressed. Should be a condition added to:

https://github.com/iceman1001/ChameleonMini-rebooted/blob/d044bb42241720f21e59690b012056fd4140109a/Firmware/ChameleonMini/Application/MifareUltralight.c#L200

wc7086, 2021-07-28:
Why hasn't this mode been added yet? What caused this mode not to be added so far?

Mobile APP can also operate this firmware. I found a good APP, but it lacks active development, does not support the latest firmware data upload and download, and has few functions.
https://github.com/kgamecarter/ChameleonMiniApp

iceman1001, 2021-07-28:
I think you have gotten things confused.  
This is a open source project,   where volunteers contribute.  
This repo don't have any active contributors.

So if you want this to be implemented, feel free to contribute

wc7086, 2021-07-28:
> I think you have gotten things confused.
> This is a open source project, where volunteers contribute.
> This repo don't have any active contributors.
> 
> So if you want this to be implemented, feel free to contribute

okay, I need some time to understand these items and related knowledge.