iceman1001/ChameleonMini-rebooted #170: [REQ] Synchronize with emsec/ChameleonMini

X-Ryl669, 2019-11-14:
How hard will it be to sync the codebase from emsec/ChameleonMini's repository?
They seems to have a lot of additional features, like:
- Support for capturing and emulating ISO15693, EM4233, TI's TagIt etc
- Lot of bug fixes

Having support for ISO15693 is really lacking here, IMHO

iceman1001, 2019-11-14:
The repo is a RevE firmware,  and since the emsec's main repo is the RevG firmware.    Which is a complete different hardware.

You find emsec's RevE firmware  https://github.com/emsec/ChameleonMini/tree/master/RevE  as a subfolder.  I don't think anyone is interested in merging those two. 

IMHO,  feel free to contribute.

closed 2019-11-14 by iceman1001

X-Ryl669, 2019-11-14:
Ok, then from what I've read, it seems like RevE hardware is able to deal with ISO15693. Would it be possible to replicate the changes made in the emsec/revG Application in order to get support for such tags ?

iceman1001, 2019-11-14:
Yup,  most definitely possible to add ISO15693 to this repo.