iceman1001/ChameleonMini-rebooted #172: [BUG] Configuring dfu-programmar

CheckMatte2, 2019-11-28:


CheckMatte2, 2019-11-28:
okay sorry im new to this site and i messed up and so here is my issue below

When i try to configure the dfu-programmar with ./bootstrap.sh as the instructions said i get this error ./bootstrap.sh: 2: aclocal: not found
./bootstrap.sh: 3: autoheader: not found
./bootstrap.sh: 4: automake: not found
./bootstrap.sh: 5: autoconf: not found
To install tab completion run
 $ sudo mv dfu_programmer /etc/bash_completion.d/
To use tab completion in this terminal run
 $ source dfu_programmer

 if anyone knows the issue and is able to help that would be much apprecitated!

leegarrett, 2019-12-30:
Re-read the instruction, you forgot to install several packages that contain these programs.

securechicken, 2020-02-23:
And this is not the correct place to troubleshot dfu-programmer related issues.

closed 2020-02-23 by securechicken