iceman1001/ChameleonMini-rebooted #175: [FORUM] Identical dump data from tag not readable

dealaddict, 2019-12-11:
I have exact dump copy 1 to 1. It's a simple Ultralight tag with just the uid info and a url address in data. I matched both nfc tag and chameleon on nxp tag info android app and it is identical. 

I can also make working tag with nfc tools pro by using "import records from tag" and writing to a blank nfc tag. This also works. But my chameleon isn't able to work/emulate it. 

I tried with Hex it, chameleongui dump editing etc. 

What must I be doing wrong?

iceman1001, 2019-12-11:
Depends on where you got your dump to start with.
For the GUI to load it, it has to be of a certain format.  Usually a dump from the proxmark3.

If you can link to your dump file, we can test an see whats going on.

dealaddict, 2019-12-11:
> Depends on where you got your dump to start with.
> For the GUI to load it, it has to be of a certain format. Usually a dump from the proxmark3.
> 
> If you can link to your dump file, we can test an see whats going on.

@iceman1001 Please check your email. I sent you the dump. The original dump was from Acr122u. As you can see in the dumps everything is identical. I've tried removing that "34"from hex edit but with or without it also its the same issue. My cloned tags with Nfc tools pro work fine but I can't use the same uid. I am pretty sure chameleon should be able to handle it. I am surely doing something wrong.

Appreciate your help.

iceman1001, 2019-12-12:
The **NxpToolsScan.txt** is just a normal .eml file.
Rename the extension to **eml** instead of **txt**  and the GUI will load it just fine with drop and drag. 
The 04xxxx.bin loads just fine. 

The binary file is'nt the same UID as the one from **NxpToolsScan.txt**

One problem that confuses ppl and that is that the GUI is always "offline" to the device.
Meaning if you set dropdown boxes,  but not selected the slot with checkbox and pressed **apply**,  the device has the old config.

When we download,  we ask the device about what size it uses,  and then we expect that much data to come from the device.  This will generate different binary files....

dealaddict, 2019-12-12:
@iceman1001 thanks. Yeah I followed the above. Out of 15 attempts it only worked once. Rest it won't emulate correctly. I think the issue is probably what @bogiton once mentioned in issue #34 if I recall. Somehow the other device is able to detect it but it doesn't do the same for a cloned plastic tag.

dealaddict, 2019-12-16:
I've been trying the entire weekend. It only worked once. I didn't even change anything in the bin file but it stopped working. I even updated to the latest firmware. @iceman1001 Do you know what causes the issue as discovered by @bogiton

iceman1001, 2019-12-16:
Was your tag a MFU  7b UID ?   That seem to have issue when being loaded with the gui.

dealaddict, 2019-12-17:
> Was your tag a MFU 7b UID ? That seem to have issue when being loaded with the gui.

@iceman1001 yes. Is there any solution? 

[image]
[image]

closed 2019-12-25 by dealaddict