iceman1001/ChameleonMini-rebooted #177: Reader attack mifare ultralight C

iNeedHelpX, 2019-12-24:
Hey wondering how this works? Do I need to dump the card first? Or do I just put in my cards uid? I need to get the keys from the reader. Problem is my card only shows 6/40 sectors. The rest need keyd

iceman1001, 2019-12-24:
reader attack Mifare UL?  Can you explain what you mean more precisly?

iNeedHelpX, 2019-12-25:
I have a mifare ultralight C card. I need the keys

iceman1001, 2019-12-25:
Thats gonna be hard.  No current solution to get UL-C keys exists.
You can sniff all the trafic and use it.

iNeedHelpX, 2019-12-25:
[image]

iNeedHelpX, 2019-12-25:
[image]
[image]

iceman1001, 2019-12-25:
As mentioned the chameleon mini can't solve your problem. 

Closning this one.

closed 2019-12-25 by iceman1001