iceman1001/ChameleonMini-rebooted #178: [FORUM] Chameleon freezing with solid light

dealaddict, 2019-12-27:
Detects in Gui but freezes with only 6 slots showing and keeps processing

Only 1 slot light is on. Can't change slot.

Won't go into dfu bootloader tried the black button trick 

Rebooted Gui also can't fully load it to force it into bootloader

I tried rescuing it with Avrisp adapter by erasing and reloading both eep and hex files. Still same issue after next reconnection. 

Any suggestions?

iceman1001, 2020-01-26:
Seems like last release of the firmware made the chameleon unstable.
I have removed that offending release,  but nevertheless this indicates that the currect source code isn't stable either. @grspy

securechicken, 2020-02-23:
You may have bricked it. Please try [this wiki page](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Getting-started#no-more-light-go-back-to-stock).
If that does not work, you will want to follow the AVRISP reset [described here](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/How-to-program-with-AVRISP-mkII), using repo provided `ChameleonMiniRDV2.0_ATxmega32A4U.hex` as Flash image in programmer, and then fully running [this wiki page](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Getting-started#no-more-light-go-back-to-stock) again.

closed 2020-02-23 by securechicken

iceman1001, 2020-02-23:
Imagine that...  someone changed their account...

securechicken, 2020-02-23:
@iceman1001 just the name, to match with other accounts on the Internet!

iceman1001, 2020-02-23:
if it wasn't for your avatar, I would have had a harder time recognize you ;)