iceman1001/ChameleonMini-rebooted #179: [REQ] Mifare DESFire emulation support?

ninjaspl0it, 2020-02-01:
Hey,

when I bought this from lab401, they claim on their website that you can emulate DESFire with this, with a custom firware. However, it doesn't seem that firmware has been created...
I've been searching online in the hopes of finding someone who has already done this, but it seems it's not yet been created. Following some old threads on the other versions there is requests dating back to 2017.

Is there any chance or possibility of emulating DESFire cards with the Chameleon Mini Rebooted as of now, or does it require someone sitting down and porting the available code to C for embedded devices?

thanks

MrNonoss, 2020-02-10:
I'm in the same situation as you and would also like to emulate desfire. 
I send an email to lab401 support to get more informations. I'll keep you updated.

iceman1001, 2020-02-10:
Claims vs reality...   "can enumlate"  doesnt equal "is implemented and works out of the box".   LAB401 doesn't develop anything in regards to firmware.

What options do you have? 
You got this open source project where you can implement the DESFIRE protocol. The chameleon already support ISO1443a,  so you don't have to do the hard stuff. :)

You can also look at the offical RevE / RevG opensource project and see if they have better support for DESFIRE already and see if you can merge it into this repo.  (Offical repo == Emsec )

MrNonoss, 2020-02-10:
> "can enumlate" doesnt equal "is implemented and works out of the box"
At least, they say it's possible but with a custom firmware ^^

I have started to wach the emsec repo, but they stated that it' for the RevE / RevG device, not the rebooted one. So.... I have a lot to learn and a lot to try here ;)

iceman1001, 2020-02-10:
with custom firmware that doesnt exists...

MrNonoss, 2020-02-13:
As promised, here is the answer from LAB401.
Spoiler alert, the short answer is "it can be done, but we do not deal with that"

> Hi,
> 
> Thank you for contacting us.
> 
> The hardware is capable of DESFire emulation, and there are some works in progress on adding software support for DESFire.
> However the project is not maintained by Lab401, so it's out of our control.
> 
> I'd suggest checking in on the issues:
> https://github.com/emsec/ChameleonMini/issues/31
> and the work in progress
> https://github.com/dev-zzo/ChameleonMini
> 
> 
> Thank you,
> Support Team

iceman1001, 2020-02-13:
Just as mentioned before.   Those companies just sell a product,  they don't develop / maintain them.

MrNonoss, 2020-02-13:
> Just as mentioned before. Those companies just sell a product, they don't develop / maintain them.

Sure. They're doing their part.
Although they are also providing trainings, so they might provide hints.

Anyway, no much help from them. 
@ninjaspl0it we will have to handle it ourselves 😂

iceman1001, 2021-01-21:
Closing,  there is a desfire mode on the offical repo for Chameleon now.

closed 2021-01-21 by iceman1001