iceman1001/ChameleonMini-rebooted #183: Switch button freezes Chameleon

dealaddict, 2020-05-22:
My unit is having problems. Switching button / tag instantly freezes my device. It won't get recognised in Gui either. I have to repair it thru Atmel Avr isp adapter.

I have so far 
Erase device 
Erase boot 
Erase chip

Reprogrammed it with few different release EEP and HEX files but still running into the same issue. 

Also DFU mode in Gui or holding button and connecting cable isn't working either. 

The unit works forever until I switch card. All tags are ultralight 64 bytes. Nothing corrupted on those bin files. Those same files work fine in other device.

My hunch is its not programming properly? 🤔

nanderh, 2020-07-26:
Same problem here,
Flashed latest release firmware, uploaded a few dumps and was functioning fine.
Then used the switch card button and it froze, also not being recognized in Windows and cannot get into DFU mode.
Got leds 3, 5 and 7 solid lit up.

Tried flashing through the headers with avrdude. Currently don't have an AVR ISP (ordered one though) so tried using a Bus Pirate and an Arduino with ISP sketch. In both cases the chameleon atxmega is not being recognized.

Although one question/comment, in the wiki example pictures about Atmel Studio, its shown to use target voltage of 4.9V, is this correct? 
https://github.com/iceman1001/ChameleonMini-rebooted/wiki/How-to-program-with-AVRISP-mkII

nanderh, 2020-07-27:
So AVR ISP arrived and reflashed it.
Went back through closed issues and it seems that after November 7th something snuck in.
After flashing https://github.com/iceman1001/ChameleonMini-rebooted/releases/tag/untagged-f7d80732aeb494e62fba can confirm the bug is not there.

Yafsec, 2020-07-28:
Same here. Worked fine until I switched from card to a detect slot. 3 leds red, Gui Hangs, Black button boot loader mode doesn't work either. Latest pre compiled firmware. Ordered AVR ISP as well.

nanderh, 2020-07-28:
Same here switching to detect slot it froze.

Looking a the commit history, on the 27/11/2019 a bunch of changes were made to the detection functionality. Pure speculation but it could be from there.

After flashing the above listed commit, detection functionality works fine again.

iceman1001, 2020-07-28:
Just to make sure.  You are all having a **RevE Rebooted** device ?   Not a  RevG based device like rev g, revG rdv2.0,  tiny, tiny pro.

Yafsec, 2020-07-28:
RevE Rebooted Proxgrind

nanderh, 2020-07-28:
RevE Rebooted from Lab401 here

iceman1001, 2020-07-28:
@grspy .... is it something you have time to look into?

grspy, 2020-07-28:
Damn, I don't have the RevE with me... and I'll stay away from home for a while. I should be able to test it around October.
Have you guys tried out any previous releases? I guess that happens only on the latest release.

Yafsec, 2020-07-28:
Cannot test atm as it is fully and utterly bricked... ;) But will try the November release as soon as my AVR ISP is in

iceman1001, 2020-07-28:
usually,   as long as you can make the device enter bootmode,  you can flash it with a default fw.
but otherwise it sounds strange. 
To recap how to trigger this behavior,

- Upload dumps,   which sorts of dump files?
- Switch button,  is it on certain dump? 
-

nanderh, 2020-07-28:
In my case I flashed the latest firmware and I had 2 dumps uploaded, in the following slots:
1. Mifare 1K 7 byte UID
2. Mifare 1k 4 byte UID
3. MF_CLASSIC_DETECTION

I had used slot 2 several times and it was working correctly.
As soon as I changed slot to the detection it froze with 3 solid leds.

Recreated the same situation with the November 7th firmware and that did not freeze.

Yafsec, 2020-07-28:
Latest firmware as well, slot 1 MF_Detection, slot 2 Mifare 1k dump. Slot 2 selected and worked, switched to slot one to read and nothing. Connected to pc, gui hangs. Hold black button while connecting USB, nothing. 
When connected to Usb, 1 led on, when I press black while connected, 3 leds on.

nanderh, 2020-07-28:
@iceman1001 if needed I got some time to look into it, not too familiar with C though.
Got any specifics in mind to check out?

iceman1001, 2020-07-28:
WinGUI hangs?  That is for another repo.

So,  only simulating a MFC dump on different slots works?
its when going between /  or only  MF_CLASSIC_DETECT  that triggers device to be unstable?

iceman1001, 2020-07-28:
MF_CLASSIC_DETECTION mode, does it work?

nanderh, 2020-07-28:
Exactly, switching from simulating a dump to MF_CLASSIC_DETECTION triggered it.

Yafsec, 2020-08-13:
Finally got my AVR ISP in, following the flashing guide but it states:
"You can upload a full flash image (e.g. "Full.hex") like this: avrdude -p x32a4u -P usb -c avrispmkII -U flash:w:Full.hex"
But just for clarification, what is Full.hex?

ceres-c, 2020-08-13:
I flashed the original rev. G with
`avrdude -c atmelice_pdi -p x128a4u -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i`
Adapt this command to your device (different flasher and MCU)

closed 2020-10-10 by dealaddict