iceman1001/ChameleonMini-rebooted #185: [FORUM] Uploading a dump on Linux using command line

petorian, 2020-09-24:
I'm searching for some instructions on using the Linux CLI to upload a dump to the Chameleon

Currently I'm trying this:

I connect with cu to the Chameleon:
`cu -l /dev/ttyACM1 -s 19200`

I then verify if the selected slot is MF_Classic_1K.
Then I issue the command `UPLOAD`
it returns 110:WAITING FOR XMODEM

Then I use the command: `~$sx dump.bin`
It then returns: `Sending dump.bin, 8 blocks: Give your local XMODEM receive command now.`
after some time: `Xmodem sectors/kbytes sent: 0/0k`
and `Retry 0: Timeout on sector ACK`
But nothing happens... dump is not uploaded.

Does anyone have a working solution?

iceman1001, 2020-09-24:
Swing by the discord server  and the channel for chameleon,  and ask there,

petorian, 2020-09-24:
Thanks I'll have a look there!

closed 2021-01-21 by iceman1001