iceman1001/ChameleonMini-rebooted #188: The bootloader

1248680677, 2021-01-18:
I'd like to make a ChameleonMini-rebooted by myself . If I need to program the bootloader by myself ?

iceman1001, 2021-01-18:
I am not sure I understand your question.  
but if you making your own device,  you will also need to put software/firmware on it.

1248680677, 2021-01-18:
> I am not sure I understand your question.
> but if you making your own device, you will also need to put software/firmware on it.

But I have no idea how to put software/firmware on it .

1248680677, 2021-01-18:
> I am not sure I understand your question.
> but if you making your own device, you will also need to put software/firmware on it.

I have found this website "https://github.com/iceman1001/ChameleonMini-rebooted/wiki/How-to-program-with-AVRISP-mkII" , but I don't want to buy an AVRISP mkII to put software/firmware on it . Is there any other way to put software/firmware on it ？

iceman1001, 2021-01-19:
not if you are building your own.

closed 2021-01-19 by iceman1001