iceman1001/ChameleonMini-rebooted #202: [BUG] All Btn Long Function changed In All Slots When I Change Anyone

Kopernikusyeung, 2022-01-17:
## Environment

|Item|Your information|
|---|---|
|**Harware**|RevE Rebooted
|**Firmware**|Chameleon-Mini rebooted v1.3 (Iceman:9f958e9|
|**GUI**|http://www.icesql.se/download/ChameleonMiniGUI/publish.htm|
|**Slot number**|ALL|
|**Slot configuration**|ALL|
|**Dump source**|N/A|
|**Reader**|N/A|
|**Flashing environment**|Win10 Pro 20H2|
|**Flashing method**|USB,flash.bat|
|**Flash memory space**|N/A|
|**Makefile configuration**|N/A|

## Bug description
### Expected function and references

### Bug

All Btn Longp functions change In all slots when I change the function in any slot and refresh the information.

### Steps to Reproduce

1. Set the slot1 to MF_Classic_1k, and set the slot8 to MF-Classic_Detection
2. Set the Btn Long Function of the slot1 to READONLY
3. Set the Btn Long Function of the slot8 to RANDOM_UID
4. Select slot1&8 and apply the settings
5. Press refresh
6. Btn Long in all the slots changeed to READONLY

iceman1001, 2022-01-17:
feel free to come up with a fix since there is very little going on in this repo.

Licsber, 2022-05-01:
Does anybody already fix this issue now?

ca1e, 2022-11-10:
not a bug, you should check the build config from makefile that contained 'BUTTON_SETTING_GLOBAL'. report bug carfully nexttime