iceman1001/ChameleonMini-rebooted #210: [FORUM] dfu-programmer 0.7.2 fails on linux

mywalkb, 2022-11-19:
I trying to update the firmware from linux but fails with this error:

```
dfu-programmer atxmega32a4u flash --eeprom --force --suppress-validation ChameleonMini.eep.scramble
0%                            100%  Programming 0x40 bytes...
[ X  ERROR
Memory write error, use debug for more info.
```
```
dfu-programmer atxmega32a4u flash --eeprom --force --suppress-validation ChameleonMini.eep.scramble --debug 100
     target: atxmega32a4u
    chip_id: 0x2fe4
  vendor_id: 0x03eb
    command: flash
      quiet: false
      debug: 100
device_type: XMEGA
------ command specific below ------
   validate: false
   hex file: ChameleonMini.eep.scramble

atmel.c:1293: atmel_flash( 0x7ffc73928c20, 0x7ffc73928b80, true, false )
atmel.c:1158: atmel_flash_prep_buffer( 0x7ffc73928b80 )
atmel.c:1331: Flash available from 0x0 to 0x3FF (64kB p. 0 to 0), 0x400 bytes.
atmel.c:1336: Data start @ 0x0: 64kB p 0; 32B p 0x0 + 0x0 offset.
atmel.c:1340: Data end @ 0x3F: 64kB p 0; 32B p 0x1 + 0x1F offset.
atmel.c:1344: Totals: 0x40 bytes, 2 32B pages, 1 64kB byte pages.
atmel.c:1041: atmel_select_memory_unit( 0x7ffc73928c20, 1 )
atmel.c:1073: Selecting eeprom memory unit.
Programming 0x40 bytes...
atmel.c:1100: atmel_select_page( 0x7ffc73928c20, 0 )
atmel.c:1113: Selecting page 0, address 0x0.
atmel.c:1428: Program data block: 0x0 to 0x3F (p. 0), 0x40 bytes.
atmel.c:1565: __atmel_flash_block( 0x7ffc73928c20, 0x7ffc73928b80, true )
atmel.c:1527: atmel_flash_populate_header( 0x7ffc73928650, 0x0, 0x3F, true )
atmel.c:1481: atmel_flash_populate_footer( 0x7ffc73928650, 0x7ffc739286d0, 65535, 65535, 65535 )
atmel.c:1627: atmel_flash: flash data dfu_download failed.
atmel.c:1628: Expected message length of 144, got -9.
atmel.c:1434: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:360: Error writing memory data. (err -4)
```

I tested version 0.7.2 and latest from dfu-programmer without success. 
I tried on Windows and works.

OT
Github support Discussion just enable it from Settings. So we can use it as forum instead of issue.

pdarcos, 2023-01-21:
I have the exact same problem. Running Pop!_OS 22.04 LTS

Did anyone figure out a solution?

tgiachi, 2023-02-24:
Same problem here!

tbaumann, 2023-03-08:
I assumed I was supposed to expect those errors (Because --force)
My dfu-programmer is even newer
```
❯ sudo dfu-programmer atxmega32a4u erase --force
Erasing flash...  Success
Checking memory from 0x0 to 0x7FFF...  Not blank at 0x7FF1.

❯ sudo dfu-programmer atxmega32a4u flash --eeprom --force --suppress-validation ./Firmware/ChameleonMini/ChameleonMini.eep.scramble
WARNING (line 3): 0x8000 address outside valid region,
 suppressing additional address error messages.
Total of 0xFA6 bytes in invalid addressed.
WARNING: 0xFA6 bytes are outside target memory,
 and will not be written.
Hex file error, use debug for more info.

❯ sudo dfu-programmer atxmega32a4u flash --force --suppress-validation ./Firmware/ChameleonMini/ChameleonMini.hex.scramble
0%                            100%  Programming 0x5900 bytes...
[ X  ERROR
Memory write error, use debug for more info.

❯ dfu-programmer --version
dfu-programmer 1.0.0
```
But it's not waking up.

I had SNAFUed the device with the wrong files I think. So I followed instructions to unbrick it.
I scrambled `ChameleonMiniRDV2.0_ATxmega32A4U.hex` and `ITS_A_CARD.hex` (In the `./Firmware/ChameleonMini/` directory to make copy&paste easier)

pdarcos, 2023-03-24:
So no one is able to flash recent releases on linux? There must be a way to do this

ca1e, 2023-04-10:
seems like you bricked your device by flashed a big fireware, try unbrick your dev by follow the wiki: 

https://github.com/iceman1001/ChameleonMini-rebooted/wiki/How-to-program-with-AVRISP-mkII

to use a new bootloader that no need to scramble the hex anymore: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Reference-RevE-Bootloader

Smartsmurf, 2026-10-13:
dfu-programmer on linux is not working. This is due to the stock bootloader, which locks on SET_CONFIGURATION.
On Windows machines libusb-win32 turns usb_set_configuration() into a no-op when the requested configuration is already the active one, so the boot loader never sees the request. Linux/libusb-1.0 does put it on the wire, which is why plain dfu-programmer fails here.  

I re-wrote dfu-programmer in Python (to make it portable). A nice side-effect is that I also fixed the host side limitation of being unable to use the full flash range (4kb left unusable).
Background: after application erase the stock bootloaders writes some bytes into the final application flash paragraph (bytes starting 0x7FF0):

0x7FF0   TEMPSENSE0
0x7FF1   LOTNUM0
0x7FF2   TEMPSENSE1
0x7FF3   LOTNUM1
0x7FF4   LOTNUM2
0x7FF5   LOTNUM3
0x7FF6   LOTNUM4
0x7FF7   LOTNUM5

Bytes are being obfuscated with "byte[i] ^ (0x34+i))"

However, this data block aborts the "flash empty" check for the last page and is reason for the "Not blank at 0x7FF1" message.

Using my python implementation I could write until 0x7F00 (theoretically up to 0x7FF0, but I skipped that part):

  python3 ./scripts/chameleon_dfu.py --target atxmega32a4u flash-hex  ChameleonMini.hex ChameleonMini.eep
  old_driver_bootloader
  Erasing flash...  Success
  Checking memory from 0x0 to 0x7EFF...  Empty.
  0%                            100%  Programming 0x40 bytes...
  [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
  0x40 bytes written into 0x400 bytes memory (6.25%).
  0%                            100%  Programming 0x7D00 bytes...
  [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
  0x7D00 bytes written into 0x7F00 bytes memory (98.43%).
  load_success!

If there is need for my script, I can set up a pull request.

iceman1001, 2026-10-13:
sounds like a useful script

closed 2026-10-13 by iceman1001